carla-simulator/carla #3658: How to accelerate the camera rendering speed of carla?

yangyueren, 2020-12-01:
Hi, I am using carla to simulate surveillance cameras in the city. I find that the rendering speed is so slow that I want to speed it up. Below is the log.

I have 8 tesla V100 gpus on my server with more than 100GB memory. But when I used nvidia-smi command to find the memory usage of carla engine, carla only use very little memory.
I wonder how to make full use of my gpus to accelerate the rendering speed. Could someone help me with it?

nvidia-smi output:
```
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1199    C+G   .../Binaries/Linux/CarlaUE4-Linux-Shipping  1435MiB |
```

log:
```
2020-12-02 Wednesday 01:25:52: Tile 1 of 1: Rendered 0 frames; 18000 remaining (0.0 FPS)
2020-12-02 Wednesday 01:26:02: Tile 1 of 1: Rendered 0 frames; 18000 remaining (0.0 FPS)
2020-12-02 Wednesday 01:28:19: Tile 1 of 1: Rendered 0 frames; 18000 remaining (0.0 FPS)
2020-12-02 Wednesday 01:29:03: Tile 1 of 1: Rendered 1 frames; 17999 remaining (0.0 FPS)
2020-12-02 Wednesday 01:29:36: Tile 1 of 1: Rendered 2 frames; 17998 remaining (0.0 FPS)
2020-12-02 Wednesday 01:30:13: Tile 1 of 1: Rendered 3 frames; 17997 remaining (0.0 FPS)
2020-12-02 Wednesday 01:30:50: Tile 1 of 1: Rendered 4 frames; 17996 remaining (0.0 FPS)
2020-12-02 Wednesday 01:31:27: Tile 1 of 1: Rendered 5 frames; 17995 remaining (0.0 FPS)
2020-12-02 Wednesday 01:32:05: Tile 1 of 1: Rendered 6 frames; 17994 remaining (0.0 FPS)
2020-12-02 Wednesday 01:32:42: Tile 1 of 1: Rendered 7 frames; 17993 remaining (0.0 FPS)
2020-12-02 Wednesday 01:33:20: Tile 1 of 1: Rendered 8 frames; 17992 remaining (0.0 FPS)
2020-12-02 Wednesday 01:33:56: Tile 1 of 1: Rendered 9 frames; 17991 remaining (0.0 FPS)
```

Axel1092, 2021-03-08:
Hi @yangyueren, right now carla does not support multi-gpu setups by default. The simulator starts and uses only a single GPU. You could try to make use of multiple GPUs by running several instances of the server in a different GPU each which you can do with the environment variable DISPLAY (e.g. `DISPLAY=1 ./CarlaUE4.sh`) and have each instance handle a subset of cameras.
The main reason for the low framerates is the cost of transferring images rendered in the GPU to the client. The more cameras you have the long this process takes.

gouchaoer, 2021-09-29:
> Hi @yangyueren, right now carla does not support multi-gpu setups by default. The simulator starts and uses only a single GPU. You could try to make use of multiple GPUs by running several instances of the server in a different GPU each which you can do with the environment variable DISPLAY (e.g. `DISPLAY=1 ./CarlaUE4.sh`) synchronize the multiple instances with the Python API and have each instance handle a subset of cameras. The main reason for the low framerates is the cost of transferring images rendered in the GPU to the client. The more cameras you have the long this process takes.

hi, do you know how to sync multiple carla instances by Python API?

closed 2023-06-02 by stale[bot]