carla-simulator/carla #3659: GNSS inaccurate coordinates – ignoring flattening of ellipsoid

BaltashovIlia, 2020-12-01:
Hi, 

As I noticed when calculating GNSS coordinates in Carla, the flattening of WGS84 ellipsoid (f = 1 / 298.257223563) is ignored. 

Accordingly, when I use flattening of ellipsoid for the reverse transformation, I get a localization error: the error is along the y axis and increases with distance from the origin (for Town1 the error varies from 0 to 2 meters). 

I think this is the reason for the problems with GNSS for many users.  

So I believe it's worth rewriting GeoLocation.cpp to keep flattening of ellipsoid in mind, maybe using GeographicLib.

corkyw10, 2021-03-08:
@doterop could you have a look at this please?

closed 2023-06-02 by stale[bot]

vjyrshn, 2023-08-23:
@BaltashovIlia Could you find a solution to this issue? We are trying to send data from Carla to Apollo and we are finding that with increasing distance from the scene origin (which was set in the custom RoadRunner map we created), there is a y offset which keeps increasing.