carla-simulator/carla #3661: Install the Unreal plugin to export maps to carla from Roadrunner

zhaoshan2, 2020-12-02:
Setup:
Ubuntu 16.04
Unreal Engine 4.21
carla_0.9.5 built from source

I was trying to export maps from roadrunner to carla, and following the instructions https://www.mathworks.com/help/roadrunner/ug/Exporting-to-CARLA.html
After doing this step:
 _4. Rebuild the plugin b. Open the project and build the plugins by clicking Yes_, I got the prompt 
[image]
 Then I tried to run `clean && make launch` to rebuild it. But further error occurs

n file included from /home/dani/carla/Unreal/CarlaUE4/Plugins/RoadRunnerCarlaIntegration/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/RoadRunnerCarlaIntegration/Module.RoadRunnerCarlaIntegration.cpp:2:
/home/dani/carla/Unreal/CarlaUE4/Plugins/RoadRunnerCarlaIntegration/Source/RoadRunnerCarlaIntegration/Private/RoadRunnerCarlaIntegration.cpp:15:10: fatal error: 'OpenDrive/OpenDriveActor.h' file not found
#include <OpenDrive/OpenDriveActor.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[11/16] Compile CarlaUE4GameModeBase.cpp
[12/16] Compile CarlaUE4.cpp
[13/16] Compile CarlaMapGenerator.cpp
[14/16] Link (ld) libUE4Editor-Carla.so
[15/16] Link (ld) libUE4Editor-CarlaUE4.so

ERROR: UBT ERROR: Failed to produce item: /home/dani/carla/Unreal/CarlaUE4/Plugins/RoadRunnerCarlaIntegration/Binaries/Linux/libUE4Editor-RoadRunnerCarlaIntegration.so
       (see ../Programs/UnrealBuildTool/Log.txt for full exception trace)
Total build time: 145.63 seconds (Local executor: 0.00 seconds)
Makefile:200: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/daniela/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

I wonder how to fix this problem so that the roadrunner plugins could be installed

EchoYJJ, 2020-12-17:
Hi! @zhaoshan2 ,I think we meet similiar problems. Have you solved it now?
My setup: CARLA_0.9.10 Unreal Engine 4.26 (no Ubuntu)
I follow the same instructions 
[image]
After doing this step:
_Rebuild the plugin.
a.Generate the project files.
Windows® - Right-click the .uproject file and select "Generate Visual Studio project files."_
It shows the error "This project does not have any source code. You need to add C++ source files to the project from the Editor before you can generate project files"
In addition, my CarlaUE4.uproject locates in 'CarlaUE4' folder.
If I ignore the error in step a, and skip to step b _"open the project and build the plugins by clicking Yes."_
Missing CarlaUE4 Module not only includes  "RoadRunnerImporter", "RoadRunnerCarlaIntegration", and "RoadRunnerMaterials", but also includes CarlaUE4 and Carla. If I still keep going, I receive the same error as you do. _CarlaUE4 could not be complied. Try rebuilding from source manually._
[image]
[image]

If you have solved this issue, please give me some advices, thanks!

zhaoshan2, 2020-12-18:
> Hi! @zhaoshan2 ,I think we meet similiar problems. Have you solved it now?
> My setup: CARLA_0.9.10 Unreal Engine 4.26 (no Ubuntu)
> I follow the same instructions
> [image]
> After doing this step: [...]

Hi, @EchoYJJ Have you ever checked with this one?
https://github.com/carla-simulator/carla/issues/1015
It didn't work for me, but anyway you can have a try and please let me know if it works.

EchoYJJ, 2020-12-19:
Thank you for you recommdations. Actually, I met some problems in the"make PythonAPI" &"make launch" like I ask here.https://github.com/carla-simulator/carla/issues/3743
I think that's probably due to the C complier. Could you please tell me how did you select the right cl.exe provided from VS2017?

gizaysargin, 2021-01-05:
> Hi! @zhaoshan2 ,I think we meet similiar problems. Have you solved it now?
> My setup: CARLA_0.9.10 Unreal Engine 4.26 (no Ubuntu)
> I follow the same instructions
> [image]
> After doing this step: [...]

Hello I am also facing the same problem. Moreover, I have also tried several different versions of Unreal Engine 4. If you were able to solve the problem, can you explain the method? Thank you. :)

erouirior, 2021-03-04:
I am facing the same problem. Could you please tell that if you have solved this problem? Thanks!

corkyw10, 2021-03-08:
The documentation is a bit misleading with the instruction _open the project and build the plugins by clicking Yes._. Once you have generated the plugin files in step 3a. then run `make launch` in the terminal to rebuild Carla/Unreal with the plugins.

EricPolizzy, 2021-06-23:
@corkyw10 I'm facing the same problem. By right-clicking on "CarlaUE4.uproject" and running it, I face the following error:

" CarlaUE4 could not be complied. Try rebuilding from source manualy."

I was wondering how we can find a straightforward approach to import RoadRunner built map to CARLA?

EricPolizzy, 2021-06-23:
@gizaysargin I'm facing the same issue and was wondering whether you have you been able to solve this problem?

EricPolizzy, 2021-06-26:
@EchoYJJ were you able to get this to work?if yes, what was your magic trick?

corkyw10, 2021-08-05:
Please see [updated](https://carla.readthedocs.io/en/0.9.12/tuto_M_add_map_alternative/#roadrunner-plugin-import) documentation for the roadrunner plugin. If you have any problems with this documentation, please open a new github issue.

closed 2021-08-05 by corkyw10