carla-simulator/carla #3663: Remove the chmod from CarlaUE4.sh

qhaas, 2020-12-02:
A few CARLA users at my organization, as well as users here (Issue #2602), appear to be encountering the following 'error' when running 'CarlaUE4.sh' from a carla deployment where they don't have full file permissions:
```
chmod: changing permissions of '/path/to/CarlaUE4-Linux-Shipping': Operation not permitted
```
This can lead to red-herrings when the chmod isn't necessarily the problem they are experiencing.  Some users even mitigate the error by adding [a sudo](https://github.com/carla-simulator/carla/issues/2138#issuecomment-616114548), which is overkill.

This is generated [via UnrealEngine](https://github.com/EpicGames/UnrealEngine/blob/4.24.3-release/Engine/Source/Programs/AutomationTool/Linux/LinuxPlatform.Automation.cs#L130).  If there is a way via configuration file to prevent UnrealEngine by way CARLA from generating it, it would be useful to those deploying CARLA in shared directories where it isn't advisable to give users write permissions, instead of having to delete the line from the generated shell script.

corkyw10, 2021-03-08:
@qhaas Thank you for the input and patience

@bernatx would you be able to have a look at this please?

parth3153, 2021-03-09:
i am having the same issue

Saquib764, 2021-04-12:
I am able to run carla with `-opengl` tag from my terminal.

But I get a similar error while trying to execute it via Jenkins.

```
chmod: changing permissions of '/home/saquib/CARLA_0.9.10/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted

Failed to load cookie file from cookie: Permission denied

Failed to load cookie file from cookie: Permission denied

Failed to load cookie file from cookie: Permission denied

Failed to load cookie file from cookie: Permission denied
```

I don't understand the issue here.

qhaas, 2021-07-07:
Since this looks to be an upstream 'feature', I asked over in [UE Answers](https://answers.unrealengine.com/questions/1039302/disable-chmod-in-run-script.html)

aliraeisdanaei, 2023-03-03:
> I am able to run carla with `-opengl` tag from my terminal.
>
> But I get a similar error while trying to execute it via Jenkins.
>
> ```
> chmod: changing permissions of [...]

-opengl is no longer available on desktop, try -vulkan

eyabesbes, 2024-04-21:
> A few CARLA users at my organization, as well as users here (Issue #2602), appear to be encountering the following 'error' when running 'CarlaUE4.sh' from a carla deployment where they don't have [...]

Hi,
I have been facing the same issue on WSL. did you manage to resolve it ? if so, could you tell me how did you do you it ?

qhaas, 2024-09-11:
> Hi, I have been facing the same issue on WSL. did you manage to resolve it ? if so, could you tell me how did you do you it ?

You can just remove the chmod line from the generated shell script used to launch carla, after ensuring that the binary it launches has execute permissions.  Since it is generated code, you have to manually do this each time you compile it or download it.

closed 2024-10-18 by Blyron