carla-simulator/carla #3666: [Testing] Mercedes Benz and Dodge Charge

XGodina, 2020-12-03:
What must be tested?

- Check materials - Good nomenclature, are materials instanced?
- Wheels Physics - are correct?
- Car Physics - Did you see correct in Lidar?
- Set Light - All lights work? Check cast shadows and ticks.
- Vehicle Factory - Can you spawn with different colors?
- Wheels Suspension - They are correct?

XGodina, 2020-12-07:
So far tested I've seen some problems.

Vehicle Factory
The cars don't change the color when you spawn it

Lincoln2020 Lights
- The break lights there are too much glow
[image]

Mercedesccc Lights
There is no glow in the brake
[image]

ChargeCop2020/Charge
The left beam doesn't work
There isn't light inside the car
There isn't light in the mesh
[image]

Improve nomenclature. We must have a better order of assets. It's not all SM_
[image]

Follow this link 
https://www.gamecoderblog.com/en/unreal-engine4/ue4-recommended-files-naming-convention

Add the good suspension - follow this link https://carla.readthedocs.io/en/latest/tuto_D_customize_vehicle_suspension/
If you have doubts about this ask Daniel Santos

XGodina, 2020-12-08:
It seems that everything is fine. But a couple of things before to close.

1 -  I would like to know the feedback of the rest of the testers @DnDare  @DSantosO 
2 - @gsmario To have good discipline and organization, you should unify the nomenclature of "mesh Glass"
[image]
[image]
[image]

DSantosO, 2020-12-10:
The physics of the Dodge and Mercedes look good. The behavior of the Lincoln is a bit off, especially during the turns because it is very easy to lose control of the car.

gsmario, 2020-12-11:
@DSantosO i've updated the car wheels width with the new values.

XGodina, 2020-12-11:
I've updated the center of mass to improve the behavior of cars. Now don't lose control.

closed 2020-12-11 by XGodina