carla-simulator/carla #3667: Sumo Co-Simulation crashes when spawning a carla vehicle on a non-allowed lane

patmalcolm91, 2020-12-03:
When running a Sumo co-simulation (e.g. via ```run_synchronization.py```), if another carla client (e.g. ```manual_control.py```) spawns a vehicle in a location corresponding to a lane where the corresponding sumo vType is not allowed (for example a passenger vehicle on a bike lane or any vehicle on a lane with ```allow="none"```), this causes Sumo to crash without any meaningful error message.

Expected behavior:
The co-simulation module should insert the Sumo vehicle into the nearest allowed lane, or should catch this exception and print a meaningful error message.

Carla version: 0.9.9
OS: Linux

Error message:
```
Traceback (most recent call last):
  File "run_synchronization.py", line 243, in synchronization_loop
    synchronization.tick()
  File "run_synchronization.py", line 103, in tick
    self.sumo.tick()
  File "/home/tum-vt/Desktop/carla-dev/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py", line 493, in tick
    traci.simulationStep()
  File "/home/tum-vt/sumo-build/tools/traci/main.py", line 217, in simulationStep
    return _connections[""].simulationStep(step)
  File "/home/tum-vt/sumo-build/tools/traci/connection.py", line 298, in simulationStep
    result = self._sendCmd(tc.CMD_SIMSTEP, None, None, "D", step)
  File "/home/tum-vt/sumo-build/tools/traci/connection.py", line 178, in _sendCmd
    return self._sendExact()
  File "/home/tum-vt/sumo-build/tools/traci/connection.py", line 88, in _sendExact
    raise FatalTraCIError("connection closed by SUMO")
traci.exceptions.FatalTraCIError: connection closed by SUMO

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run_synchronization.py", line 316, in <module>
    synchronization_loop(arguments)
  File "run_synchronization.py", line 256, in synchronization_loop
    synchronization.close()
  File "run_synchronization.py", line 222, in close
    self.sumo.destroy_actor(sumo_actor_id)
  File "/home/tum-vt/Desktop/carla-dev/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py", line 446, in destroy_actor
    traci.vehicle.remove(actor_id)
  File "/home/tum-vt/sumo-build/tools/traci/_vehicle.py", line 1569, in remove
    self._setCmd(tc.REMOVE, vehID, "b", reason)
  File "/home/tum-vt/sumo-build/tools/traci/domain.py", line 185, in _setCmd
    self._connection._sendCmd(self._cmdSetID, varID, objectID, format, *values)
  File "/home/tum-vt/sumo-build/tools/traci/connection.py", line 178, in _sendCmd
    return self._sendExact()
  File "/home/tum-vt/sumo-build/tools/traci/connection.py", line 83, in _sendExact
    self._socket.send(length + self._string)
AttributeError: 'Connection' object has no attribute '_socket'
```

corkyw10, 2021-03-08:
@Axel1092 are you able to have a look at this please?

joel-mb, 2021-03-16:
Hi @patmalcolm91!
I suspect this can be related to the fact that CARLA vehicles are spawned in SUMO using always the same route. If the spawned vehicle is not allowed to be spawned at that route it may crash. However, I haven't been able to reproduce the issue with the CARLA maps. Are you using a custom map or a modified version of the `vtypes,json` file?

patmalcolm91, 2021-03-16:
Hi, @joel-mb. I should have specified that I'm using a custom map when encountering this issue. To create the map, we started with a custom SUMO net file, generated street geometry from it, and used a dummy XODR (since we don't need the CARLA internal navigation at all for our use case). The SUMO network consists of lanes with various vehicle type allowances (some with ```allow="all"```, some allowing only passenger vehicles, some allowing only bicycles, etc.). I first encounted the issue when trying to manually spawn a bicycle on a road lane with ```allow="passenger"``` (i.e. no bicycles allowed).

I imagine the reason this is difficult to recreate in the default maps is that in the accompanying SUMO net files, almost all of the lanes allow nearly all vTypes. However, I believe you should be able to recreate it if you try manually spawing a passenger vehicle at a coordinate corresponding to a sidewalk or a parking/shoulder lane (for example, the shoulder of the ring road in Town05).

patmalcolm91, 2021-07-21:
These changes appear to have resolved the issue. Thanks, @joel-mb!

PR #4430 (2021-07-26): [Co-Simulation] Create SUMO routes per vehicle class

#### Description

This PR introduces some changes to create a SUMO route per vehicles class. The vehicle won't be spawned if there isn't any allowed edge in the SUMO network.

Fixes #3667

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3.6.9
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/4430)
closed 2021-07-21 by patmalcolm91