carla-simulator/carla #3668: New vehicle import issue

kanakab, 2020-12-03:
I want to develop a new vehicle and import in CARLA,
As per CARLA guidelines https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/
I have downloaded the skeleton of the Vehicle and when I try to import the fbx file in Blender,
 I am getting the following error: ASCII fbx files are not supported

What I have tried : 
1. I opened the file seperately in VSCode and saved it as a new file

Please guide me on how to proceed further. If I am doing something wrong, please let me know

BAROBA-3d, 2020-12-04:
It will not import because Blender does not import ASCII FBX files...
But you can try this:
https://www.reddit.com/r/blender/comments/csw9as/ascii_fbx_not_importing_try_this_method_28/

gsmario, 2020-12-07:
Yes this is an issue with blender, you should try what @BAROBA-3d recommended, saving the fbx in a format that blender can open will work.

kanakab, 2020-12-07:
Thank you @gsmario , @BAROBA-3d 

I converted the file from ASCII fbx to Binary fbx using Autodesk fbx converter and it worked after that.
Here are the steps I followed: https://www.artstation.com/mattmurch/blog/ZA8l/quickly-fix-blender-ascii-fbx-import-error

closed 2020-12-07 by kanakab