carla-simulator/carla #3670: Some bounding box extents are 0 in Y axis

machar94, 2020-12-03:
Carla version number: 0.9.10.1-112-g993f440b
Commit: 993f440bb804f22ec4e94653e1dd97fb0ce6babc
Platform/OS: Ubuntu 18.04.4 LTS

I have observed that some obstacles (mostly 2 wheelers) have an extent in the Y dimension of 0. I noticed this in my Carla/Apollo integration when the bounding boxes in the visualizer appeared distant and only in 2D. I modified the spawn_npc.py script (provided below) to print out the bounding boxes of any actors that reported a value of 0 in any of the axes. In addition, with the release of 0.9.10 I saw that bounding boxes can be obtained globally from the world object. I also checked whether there were any bounding boxes from the get_level_bbs method of world that also have a 0 extent value in any of the directions and there are none! Below is the output of the modified spawn_npc.py script for 1 tick. I have called out in bold the 0 extent values and if you see the relative location for those bounding boxes they appear to be greatly off in the y axis. In addition, out of all of the printed global bounding boxes (filtered for vehicles - not sure why there are more than 10 when I requested 10 vehicles), I am able to match the actor to the bounding box with proper extents since the location of those bounding boxes are in global coordinates. Out of 10 harley davidsons requested 3/10 were defective. This proportion changes as I vary the number of requested vehicles. I know this commit is about a month behind the head, but I was hoping if someone could quickly run the spawn_npc.py script I've provided and comment on whether they observe a similar issue with the bounding box extents. The proportions at the end of the tick output reflect the number of defective bounding boxes detected with each method.

<pre>
###### Tick     4 #####
Actor ID(155): Location(x=<b>163.864212</b>, y=<b>-193.899994</b>, z=0.182794) BoundingBox(Location(x=-0.008867, y=<b>194.266083</b>, z=0.659914), Extent(x=1.177870, y=<b>0.000000</b>, z=0.638285), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000)))
Actor ID(151): Location(x=<b>-145.519318</b>, y=<b>21.290268</b>, z=0.182794) BoundingBox(Location(x=-0.008867, y=<b>145.874252</b>, z=0.659914), Extent(x=1.177870, y=<b>0.000000</b>, z=0.638285), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000)))
Actor ID(149): Location(x=<b>150.919495</b>, y=<b>-162.625595</b>, z=4.425494) BoundingBox(Location(x=-0.008857, y=<b>148.434616</b>, z=0.659914), Extent(x=1.177870, y=<b>0.000000</b>, z=0.638285), Rotation(pitch=0.000000, yaw=0.000000, roll=0.000000)))
BoundingBox(Location(x=-112.199997, y=-95.554062, z=0.819803), Extent(x=2.513388, y=1.075773, z=0.817764), Rotation(pitch=0.000000, yaw=-90.000114, roll=0.000000))
BoundingBox(Location(x=-118.697632, y=-95.184227, z=0.965306), Extent(x=1.993842, y=0.925424, z=0.808555), Rotation(pitch=1.186370, yaw=-89.999992, roll=0.000000))
BoundingBox(Location(x=-121.698212, y=-95.293007, z=0.975657), Extent(x=2.678740, y=1.016602, z=0.707373), Rotation(pitch=0.593338, yaw=-90.000092, roll=0.000000))
BoundingBox(Location(x=-104.099998, y=-94.498207, z=0.447562), Extent(x=1.902900, y=0.985138, z=0.737515), Rotation(pitch=1.274575, yaw=-90.000053, roll=0.000000))
BoundingBox(Location(x=-194.599777, y=104.801025, z=0.753248), Extent(x=1.933110, y=0.952598, z=0.938968), Rotation(pitch=0.000000, yaw=0.000161, roll=0.000000))
BoundingBox(Location(x=-194.609406, y=101.900002, z=0.833576), Extent(x=2.238886, y=1.034658, z=1.018896), Rotation(pitch=0.000000, yaw=0.000155, roll=0.000000))
BoundingBox(Location(x=-188.957382, y=105.000153, z=0.740486), Extent(x=2.302255, y=0.965797, z=0.927423), Rotation(pitch=0.000000, yaw=-0.000092, roll=0.000000))
BoundingBox(Location(x=-188.886032, y=101.900002, z=0.550788), Extent(x=1.902900, y=0.985138, z=0.737515), Rotation(pitch=0.000000, yaw=0.000155, roll=0.000000))
BoundingBox(Location(x=-19.074694, y=154.678558, z=0.950356), Extent(x=2.305503, y=1.120857, z=0.833638), Rotation(pitch=0.000000, yaw=90.000114, roll=0.000000))
BoundingBox(Location(x=-24.814787, y=155.477249, z=0.789589), Extent(x=1.902900, y=0.985138, z=0.737515), Rotation(pitch=0.000000, yaw=90.000114, roll=0.000000))
BoundingBox(Location(x=73.789474, y=154.503555, z=1.257309), Extent(x=1.933110, y=0.952598, z=0.938968), Rotation(pitch=0.000000, yaw=-90.000114, roll=0.000000))
BoundingBox(Location(x=31.801071, y=113.743866, z=1.077602), Extent(x=1.902900, y=0.985138, z=0.737515), Rotation(pitch=0.000000, yaw=90.000114, roll=0.000000))
BoundingBox(Location(x=149.546371, y=149.862900, z=2.071619), Extent(x=2.096415, y=0.908093, z=0.736916), Rotation(pitch=0.000000, yaw=90.000160, roll=-2.444794))
BoundingBox(Location(x=130.541885, y=79.543892, z=1.270044), Extent(x=2.302255, y=0.965797, z=0.927423), Rotation(pitch=0.000000, yaw=-90.000114, roll=0.000000))
BoundingBox(Location(x=188.755371, y=87.799812, z=1.325878), Extent(x=2.302255, y=0.965797, z=0.927423), Rotation(pitch=0.000000, yaw=-90.000114, roll=0.000000))
BoundingBox(Location(x=188.770676, y=80.130074, z=1.118880), Extent(x=2.513388, y=1.075773, z=0.817764), Rotation(pitch=0.000000, yaw=-90.000114, roll=0.000000))
BoundingBox(Location(x=89.850563, y=-222.489227, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=30.000006, roll=0.000000))
BoundingBox(Location(x=89.805458, y=-222.514954, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=30.000006, roll=0.000000))
BoundingBox(Location(x=84.465660, y=-221.834503, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=144.999985, roll=0.000000))
BoundingBox(Location(x=84.508026, y=-221.864532, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=144.999985, roll=0.000000))
BoundingBox(Location(x=79.858749, y=-221.939240, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=44.999939, roll=0.000000))
BoundingBox(Location(x=79.821823, y=-221.975739, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=44.999939, roll=0.000000))
BoundingBox(Location(x=158.934143, y=-222.489227, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=30.000006, roll=0.000000))
BoundingBox(Location(x=158.889038, y=-222.514954, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=30.000006, roll=0.000000))
BoundingBox(Location(x=153.549240, y=-221.834503, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=144.999985, roll=0.000000))
BoundingBox(Location(x=153.591614, y=-221.864532, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=144.999985, roll=0.000000))
BoundingBox(Location(x=148.942337, y=-221.939240, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=44.999939, roll=0.000000))
BoundingBox(Location(x=148.905411, y=-221.975739, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=44.999939, roll=0.000000))
BoundingBox(Location(x=-162.268234, y=-3.965599, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=-20.000027, roll=0.000000))
BoundingBox(Location(x=-162.316925, y=-3.947565, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=-20.000027, roll=0.000000))
BoundingBox(Location(x=-162.596848, y=0.193104, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=24.999521, roll=0.000000))
BoundingBox(Location(x=-162.644043, y=0.171425, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=24.999521, roll=0.000000))
BoundingBox(Location(x=-188.866211, y=-3.732635, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=-0.000458, roll=0.000000))
BoundingBox(Location(x=-188.918121, y=-3.732342, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=-0.000458, roll=0.000000))
BoundingBox(Location(x=-197.405548, y=-0.344727, z=0.786582), Extent(x=2.487122, y=1.019201, z=0.777148), Rotation(pitch=0.000000, yaw=-0.000458, roll=0.000000))
BoundingBox(Location(x=-197.457458, y=-0.344434, z=0.950204), Extent(x=2.267992, y=0.926515, z=0.609083), Rotation(pitch=0.000000, yaw=-0.000458, roll=0.000000))
[... cut ...]
        for i in range(0, len(all_id), 2):
            all_actors[i].stop()

        print('\ndestroying %d walkers' % len(walkers_list))
        client.apply_batch([carla.command.DestroyActor(x) for x in all_id])

        time.sleep(0.5)

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```

How I run the spawn_npc.py script.
``` python
python3 spawn_npc.py --host [host] -n 10 -w 0 --filterv vehicle.harley-davidson.low_rider 2>&1 | tee output.txt
```

jamesgunnfiveai, 2020-12-10:
Commit: bdeafba

I'm observing the same behaviour. Similar to what you've observed, this only affects some spawned agents of particular classes. The problematic classes I've encountered are:
```
vehicle.diamondback.century
vehicle.harley-davidson.low_rider
vehicle.gazelle.omafiets
vehicle.kawasaki.ninja
vehicle.yamaha.yzf 
```

I believe all of these are two-wheelers.

ARSekkat, 2021-01-03:
I had the same issue with the last release 0.9.11, also just with 2Wheeled.

corkyw10, 2021-03-08:
@doterop would you be able to have a look at this please?

rossivalen, 2021-03-29:
observed the same issue, i was thinking about assigning a fixed value for that dimension (since all pedestrians have the same extents values) and solve that way.
Would you mind sharing any alternative solution you found, if any? Thanks

100rab-S, 2021-12-03:
This issue is still persistent with 0.9.13 version too. It is only with 2-wheelers. Due to this y = 0 extent, the bounding box could not be plotted when we project the 3D bounding box to 2D pixel wise bounding box.

yli-miovision, 2022-10-04:
Saw the same issue in 0.9.13. Any suggested workaround like what @rossivalen mentioned?

qiuhaining, 2022-12-16:
I had the same issue with the last release 0.9.13, also just with 2Wheeled.

qiuhaining, 2023-04-17:
收到了！谢谢！

carlyangdong, 2023-08-25:
> Hi, We are checking the bug. It seems that the bounding of the bike itself is correct, but when we try to add the bounding of the rider, it is done incorrectly. We are fixing that.
> 
> Thanks for warning about that.

so, which version can be used?

JiahaoXia, 2023-11-15:
Same for me using version 0.9.14. Does anyone know how to fix this?

MMNavetty, 2024-01-25:
Please find a solution to this problem. Does anyone checks if it is still the same on 0.9.15 ?

zhang-ruining, 2024-03-29:
carla 0.9.15 is fine

MarcelPiNacy-CVC, 2024-05-10:
Hi,

as of 0.9.15 does anyone still have issues with this?

Marcel

zhang-ruining, 2024-05-10:
0.9.15 no issues

closed 2024-05-15 by MarcelPiNacy-CVC

CallMeFrozenBanana, 2024-09-26:
> Please find a solution to this problem. Does anyone checks if it is still the same on 0.9.15 ?

Yes, I think there is the same problem on v0.9.15