carla-simulator/carla #3671: Running Carla Offscreen Linux

Stephanehk, 2020-12-04:
I am attempting to run Carla offscreen on Linux after installing it using Option B (package installation). I have tried to following commands but continue to get a signal 11 error (ie: I do not think the Carla simulator is actually starting offscreen). Is this because I am not using the Docker installation? I would appreciate any help, thanks.

Commands tried:
`DISPLAY= ./CarlaUE4.sh -opengl`
`SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh`

Error:
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=99968 LargeMemoryPoolOffset=231104 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

closed 2020-12-05 by Stephanehk

heethesh, 2020-12-17:
how did you resolve this?

qhaas, 2021-01-26:
If you are wanting to run headless and just dump data to files (i.e. no desire for an X-session), we use `SDL_VIDEODRIVER=offscreen carla/CARLA_0.9.11/CarlaUE4.sh -opengl`, works in RHEL7, without containers on our nvidia GPUs.  Note the [`-opengl` flag](https://carla.readthedocs.io/en/latest/adv_rendering_options/#deprecated-emulate-the-virtual-display):
```
This is only possible in Linux while using OpenGL. Unreal Engine crushes when Vulkan is running off-screen, and this issue is yet to be fixed by Epic.
```

heethesh, 2021-01-27:
I finally decided to run it from a Docker container, which was the only solution that worked for me on Ubuntu 18/20