carla-simulator/carla #3673: Understanding vehicle velocities

danielaDESM, 2020-12-04:
Hi!,

I am using CARLA 0.9.10.1 (release version) and I am recording data from different sensors, namely GNSS, IMU and LiDAR, to process them with other developments tools I am working on. I am also logging information from the vehicle, which I am planning to use as ground truth, but they do not make too much sense to me. The configuration is quiet simple, I am only working in a TimeFixed-Synchronized modality in CARLA and the vehicle is set in autopilot. The sensors are spawned and so far they seem to retrieve correctly the information, specially in the case of LiDAR and GNSS. There are things that I still do not like from the values I get from the IMU, but that is another topic. 
The velocities I get from the vehicle are mostly zero in the X and Y axes and for the Z axes they report values between 4 -6 m/s, which should not be the case. The car is driving forward and, according to the defined CS of CARLA-UE4, where X-axis is pointing forward, Y-axis to the right and Z-axis is up, I should be getting values mainly (if not only) in X- and Y-axis. 

Could someone clarify me if it is a known issue? Is it the autopilot?

I am quiet interested in having the velocity stated in the three components and avoid using the function that gives you the forward speed.

Thanks in advance!

Operative system: Windows

corkyw10, 2021-03-08:
@DSantosO could you have a look at this please?

DSantosO, 2021-03-10:
Hello @danielaDESM,

I have tried to reproduce your problem but I cannot see it. I have tried with the current dev and the package you mentioned.
For current dev this is the information of a car (Location, Forward Direction, Velocity, Velocity in local coordinates, Acceleration and IMU acceleration) for 4 steps after setting a velocity of 20m/s :

> Step: 45497 1066.966
>   Location:     +227.310   -3.124   +0.002
>   Forward:        -0.000   +1.000   -0.000
>   Velocity:       -0.000  +19.542   +0.001
>   LocVeloc:      +19.542   -0.000   +0.008
>   Acceleration:   -0.000 +195.421   +0.013
>   IMU  Acceler: +196.133   -0.000   +9.899
>
> Step: 45498 1067.066
>   Location:     +227.310   -1.172   +0.002
>   Forward:        -0.000   +1.000   -0.000
>   Velocity:       -0.000  +19.505   +0.001
>   LocVeloc:      +19.505   +0.000   +0.006
>   Acceleration:   -0.000   -0.368   -0.004
>   IMU  Acceler:   -0.917   +0.000   +9.795
>
> Step: 45499 1067.166
>   Location:     +227.310   +0.777   +0.002
>   Forward:        -0.000   +1.000   -0.000
>   Velocity:       -0.000  +19.469   +0.001
>   LocVeloc:      +19.469   +0.000   +0.003
>   Acceleration:   -0.000   -0.361   -0.004
>   IMU  Acceler:   -0.364   -0.002   +9.806
>
> Step: 45500 1067.266
>   Location:     +227.310   +2.722   +0.002
>   Forward:        -0.000   +1.000   -0.000
>   Velocity:       -0.000  +19.433   +0.000
>   LocVeloc:      +19.433   +0.000   +0.002
>   Acceleration:   +0.000   -0.360   -0.003
>   IMU  Acceler:   -0.363   +0.003   +9.808

How are you moving the vehicle? Have you disabled the gravity or the simulation of physics?

danielaDESM, 2021-03-30:
I am moving the vehicle through waypoints: waypoint.next(). So, in autopilot I do not have this problem. It is only when I move through waypoints. Is it not available the imu information in this modality?  Here the code block:

`print('Advancing in simulation')
        # Create a synchronous mode context.
        with CarlaSyncMode(world, lidar, imu, gnss, fps=1/delta) as sync_mode:
            # Create queues for sensors to store data based on the operating rate
            lidar_queue =queue.Queue(maxsize=lidar_queue_size)
            while True:
                # Advance the simulation and wait for the data.
                snapshot, cloud, imuMeas, gnssMeas = sync_mode.tick(timeout=4.0)

                #---- Vehicle navigation ----
                traffic_manager.ignore_lights_percentage(vehicle, 100)
                if vehicle_navigation==1:
                    # Choose the next waypoint and update the car location.
                    waypoint = random.choice(waypoint.next(0.5))
                    vehicle.set_transform(waypoint.transform)
                elif vehicle_navigation==2:
                    vehicle.set_autopilot(True)    
                # Save cloud in PCD
                if not lidar_queue.full():
                    lidar_queue.put(cloud)
                else:
                    # Construct cloud from queue:
                    frame_number, timestamp, cloud_array, cloud1 =lidarFunc.construct_cloud_from_queue(lidar_queue)
                    save_to_PCD_from_queue(frame_number, timestamp, cloud_array)
                    lidar_queue.put(cloud)      
                    record_AllSENSORS_state(f_debug_lidar, cloud1, frame_number, timestamp) 
                # Save IMU measurements
                record_imu(f_imu, imuMeas)
                # Save GNSS measurements
                record_gnss(f_gnss, gnssMeas)
                # Save Vehicle data
                record_vehicle_state(f_vehicle, vehicle, imuMeas.frame, imuMeas.timestamp)
                record_Inte_state(f_Inte, vehicle, imuMeas.frame, imuMeas.timestamp)
                # Save sesnors debugging data 
                record_AllSENSORS_state(f_debug_imu, imuMeas, imuMeas.frame, imuMeas.timestamp)
                record_AllSENSORS_state(f_debug_gnss, gnssMeas, gnssMeas.frame, gnssMeas.timestamp)  
                # Update Spectator View
                spectator.set_transform(cam.get_transform()) `

werewolfdev, 2021-10-08:
Hi Iam trying to record the data using metrics manager and velocity values comed negative as well as the y coordinates shows up like everything is mirrored .Does anyone have any insights on this