carla-simulator/carla #3676: Pulse audio error when starting Carla server

Stephanehk, 2020-12-07:
I am attempting to start the Carla server using the command `./CarlaUE4.sh ` on Ubuntu Linux after using the precompiled installation. I get the following error, however, and then the server hangs.

`4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
xcb_connection_has_error() returned true
Failed to create secure directory (/run/user/40445/pulse): No such file or directory`

Upon hitting command c:
`^CFUnixPlatformMisc::RequestExitWithStatus
FUnixPlatformMisc::RequestExit
Exiting abnormally (error code: 130)`

Upon Googling this, it seems to be an error relating Pulse audio. I am not sure why I am getting this error message. Are there any workarounds or fixes? Is Pulse even needed to run Carla?

corkyw10, 2021-03-08:
@doterop could you follow up on this please?

doterop, 2021-03-23:
Hi @Stephanehk 

Are you getting this error only with Carla? It seems something related to the configuration of your machine.