carla-simulator/carla #3677: Multiprocessing Agent Hangs at try_spawn_actor

kareido, 2020-12-07:
Hello. I am trying to create multiple agents as subprocess and train them asynchronously using CARLA 0.9.10 environment on Ubuntu 18.04. However, I found when I call try_spawn_actor in a subprocess, it hangs forever, causing the server dead. Specifically, my agent looks like:  
```python3
import torch.multiprocessing as mp # I use pytorch 1.4 with python 3.7.9

class SomeAgent(mp.Process):
    ....

    def run(self):
        ...
        some_carla_env._world.try_spawn_actor(vehicle_blueprint, transform)
        ...

```

Then I tried to create multiple agent by:
```python3
workers = [SomeAgent(...) for i in range(10)]
[w.start() for w in workers] # using fork here
[w.join() for w in workers]
```

When I debug this problem, there is really not to much traceback I have. What I am only sure is that it never returns from try_spawn_actor thus hanging there forwever. Looking forward to any solution to this problem. Thanks.

corkyw10, 2021-03-08:
@kareido thank you for your patience, are you still facing this issue?

kareido, 2021-03-20:
> @kareido thank you for your patience, are you still facing this issue?

Hi. Thanks for the response. Upon further debugging, I found the problem is due to some line in my code calling:
`world.tick()`
which is one of CARLA's interfaces that ticks the CARLA world one timestep ahead. 

For CARLA 0.9.10 I am using, yes I am still facing this issue.

Thanks.

glopezdiest, 2021-12-20:
Closing this issue while waiting for the OP to provide more info

closed 2021-12-20 by glopezdiest