carla-simulator/carla #3683: With set_hybrid_physics_mode, get_velocity() returns (0, 0, 0) even when near to hero

chauvinSimon, 2020-12-07:
I open this issue following a discussion with @jackbart94 in the discord channel. I did not find any duplicate among the existing issues. https://github.com/carla-simulator/carla/issues/3145 could seem close, but the problem only prompts when replaying.

Use case here:
- An ego and a **second car (with TM)** are spawn close to another.
- The other cars drives and **moves**.
- We **read its speed**, and check its distance to the ego.
- -> We expect the **measured speed to be non zero** if the other moves.

To reproduce, you can run this script
- Once with `USE_HYBRID_PHYSICS = True`
- and once with `USE_HYBRID_PHYSICS = False` (line~63).

In both cases, the ego ("hero" attribute) sees the other vehicle driving close (distance to ego ~ 40m). **Its location changes, because it is moving**. But:

- with `USE_HYBRID_PHYSICS = False`: `speed != 0` (expected)

- with `USE_HYBRID_PHYSICS = True`: **`speed == 0` <- PROBLEM**

Note1: I am using `0.9.10.0` and `python3.7`.

Note2: for the ego, everything looks file. When applying `carla.VehicleControl` to it, its `get_velocity()` is not zero (as expected).

```
import carla
from dataclasses import dataclass
import logging
import numpy as np
import time
from typing import List

logging.basicConfig(level=logging.DEBUG)

@dataclass
class VehicleDescription:
    model: str
    spawn_point_6d: List[float]

if __name__ == "__main__":
    client = carla.Client('127.0.0.1', 2000)
    world = client.load_world("Town02")

    ego = VehicleDescription(
        model="vehicle.nissan.micra",
        spawn_point_6d=[
            27.72001838684082,  # + 7
            187.9199981689453,
            0.5,
            0.0,
            -179.9996337890625,
            0.0]
    )

    other_vehicles = [
        VehicleDescription(
            model="vehicle.nissan.patrol",
            spawn_point_6d=[
                27.72001838684082 + 60,  # + 0
                187.9199981689453,
                0.5,
                0.0,
                -179.9996337890625,
                0.0])
    ]

    # spawn ego
    blueprint_library = world.get_blueprint_library().filter(ego.model)
    blueprint = blueprint_library[0]
    blueprint.set_attribute("role_name", "hero")
    transform = carla.Transform(
        carla.Location(
            *ego.spawn_point_6d[:3]
        ),
        carla.Rotation(
            *ego.spawn_point_6d[3:]
        )
    )
    ego_vehicle = world.try_spawn_actor(blueprint, transform)

    traffic_manager = client.get_trafficmanager(port=8000)

    # TODO: here is the problem!
    # set_hybrid_physics_mode mode is meant to reduce veh. dynamics computation: far away veh are just teleported
    # But it seems to make get_velocity()=(0, 0, 0) for all cars, not just for those which are far away
    USE_HYBRID_PHYSICS = True
    traffic_manager.set_hybrid_physics_mode(USE_HYBRID_PHYSICS)

    traffic_manager.set_hybrid_physics_radius(70.0)
    traffic_manager.set_synchronous_mode(True)

    vehicles_list = []
    batch = []
    for veh in other_vehicles:
        blueprint = world.get_blueprint_library().filter(veh.model)[0]
        blueprint.set_attribute('role_name', 'autopilot')
        transform = carla.Transform(
            location=carla.Location(
                x=veh.spawn_point_6d[0],
                y=veh.spawn_point_6d[1],
                z=veh.spawn_point_6d[2]
            ),
            rotation=carla.Rotation(
                pitch=veh.spawn_point_6d[3],
                yaw=veh.spawn_point_6d[4],
                roll=veh.spawn_point_6d[5],
            )
        )
        batch.append(carla.command.SpawnActor(blueprint, transform).then(
            carla.command.SetAutopilot(carla.command.FutureActor, True, traffic_manager.get_port())))

    for response in client.apply_batch_sync(batch, True):
        if response.error:
            logging.error(response.error)
        else:
            vehicles_list.append(response.actor_id)

    actors_list = []
    for actor_id in vehicles_list:
        a = world.get_actor(actor_id=actor_id)
        assert a.id == actor_id
        traffic_manager.ignore_lights_percentage(a, 100.0)
        actors_list.append(a)

    logging.info('{} vehicle(s) were spawned'.format(len(vehicles_list)))

    def print_veh_infos():
        ego_loc = ego_vehicle.get_location()
        for vehicle in world.get_actors().filter('vehicle.*'):
            if vehicle.id != ego_vehicle.id:
                velocity = vehicle.get_velocity()
                other_loc = vehicle.get_location()
                logging.info("vehicle {} has speed={:.2f} at {:.2f}, {:.2f}".format(
                    vehicle.id,
                    np.linalg.norm(np.array([velocity.x, velocity.y])),
                    other_loc.x,
                    other_loc.y
                ))
                logging.info("distance to ego = {:.2f}m".format(
                    ego_loc.distance(other_loc)
                ))

    for _ in range(100):
        print_veh_infos()
        time.sleep(0.1)
        world.tick()

    for actor in actors_list + [ego_vehicle]:
        if actor.is_alive:
            logging.info("destroying {}".format(actor.id))
            actor.destroy()

```

corkyw10, 2021-03-08:
@jackbart94 Could you follow up on this please?

jackbart94, 2021-04-12:
Took me a while to answer, but here's the solution: you must set hybrid mode before spawning the hero vehicle. I've done so and it works flawlessly, so I'll close this.

closed 2021-04-12 by jackbart94

qhaas, 2021-05-19:
> Took me a while to answer, but here's the solution: you must set hybrid mode before spawning the hero vehicle. I've done so and it works flawlessly, so I'll close this.

That should be in the documentation, I just ran into this and was wondering why my vehicles were moving at 0 speed.