carla-simulator/carla #3684: Cannot load Town06/07 and Town10 map

heethesh, 2020-12-07:
Version: 0.9.10.1
OS: Ubuntu 20.01
Built from source

I'm unable to import Town10. Looks like the HD map data is not present for Town10. I did run the `Update.sh` and `ImportAssets.sh` and can load the maps of Town01-05. Although Town06.pcd and Town07.pcd files are present in the HDMaps folder, the client throws the exception: `map not found`. Is there something that I missed in the docs to load newer maps?

jldez, 2020-12-10:
You also have to download the maps. See for exemple: 
"AdditionalMaps_0.9.10.tar.gz"
https://github.com/carla-simulator/carla/releases/tag/0.9.10

I'm not quite sure of the procedure from there, but I think you have to place the maps in the Import directory and then run ImportAssets.sh

heethesh, 2020-12-10:
@jldez Since I'm building from source, I assumed I would have all the assets in the repository, if not, this should probably be documented somewhere. I'll try what you suggested.

jldez, 2020-12-11:
Did it work? I would like to try it myself.

heethesh, 2020-12-14:
@jldez yes it did, I downloaded the `tar.gz` additional maps package and placed the file in `Dist/CARLA_Shipping_0.9.10.1-dirty/LinuxNoEditor/Import` followed by running `./ImportAssets.sh`. Also found some documentation related to this [here](https://carla.readthedocs.io/en/latest/tuto_A_create_standalone/). I can load maps Town06 and Town07, still unable to load Town10 though.

heethesh, 2020-12-14:
Okay, I'm now able to import Town10 as well, the name of the map had to be `Town10HD` not `Town10`.

closed 2020-12-14 by heethesh

EthanCodesss, 2021-01-24:
Hello, I have the same problem, what should I do on windows?