carla-simulator/carla #3686: How to set geoReference of the xodr exported by osm?

yangyueren, 2020-12-08:
Hi:
I am importing a real-world map into carla from OpenStreetMap following https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/ . Below is the map:
[image]

I converted the osm into xodr, but I find that there is no geoReference tag in the transformed xodr. In the PythonAPI/util/opendrive folder, I find a file named TownBig.xodr, which contains geoReference tag.

TownBig.xodr:
` <header revMajor="1" revMinor="4" name="" version="1" date="2020-04-13T17:23:35" north="1.2341304024654144e+3" south="-1.5044203887070851e+3" east="7.6625546144109308e+2" west="-1.2676734634767433e+3" vendor="VectorZero">
        <geoReference><![CDATA[+proj=tmerc +lat_0=37.35429341239328 +lon_0=-122.0859797650754 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>`

my own xodr:
`<header revMajor="1" revMinor="4" name="" version="1.00" date="Tue Dec  8 19:10:44 2020" north="9656.30" south="0.00" east="12505.37" west="0.00"/>`

I try to change the header of my own xodr into:
`<header revMajor="1" revMinor="4" name="" version="1" date="2020-04-13T17:23:35" north="9656.30" south="0.00" east="12505.37" west="0.00" vendor="VectorZero">
        <geoReference><![CDATA[+proj=tmerc +lat_0=40.7373 +lon_0=-73.9929 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>
        <userData>
            <vectorScene program="RoadRunner" version="2019.2.11 (build 1eb1d9099)"/>
        </userData>
    </header>`

When I use m.transform_to_geolocation function to transform the waypoint location in carla into to a carla.GeoLocation, which represents world coordinates, I find it doesn't match the world map. Please check the image below:
[image]

Could someone help me with the header content of xdor?

ebadi, 2020-12-10:
I had a similar problem a while ago and I am curious to know more about your approach.

> I converted the osm into xodr, but I find that there is no geoReference tag in the transformed xodr. 

I assume you didn't use RoadRunner but used osm2odr tool, right?

> I try to change the header of my own xodr into:

How did you generate your xodr header?

yangyueren, 2020-12-10:
> I had a similar problem a while ago and I am curious to know more about your approach.
>
>
> I assume you didn't use RoadRunner but used osm2odr tool, right?
>
>
> How did you generate your xodr header?

I didn't use RoadRunner. I just download the osm file from openstreemap website and use the function carla.Osm2Odr to convert osm into xodr and write xodr into the file. It 
I just open the file with sublime and find that there is already header tag, but no geoReference tag inside it.
So I added the geoReference tag into the header by myself by copying the content of the geoReference tag inside TownBig.xodr in the PythonAPI/util/opendrive folder to my own xodr file (I modified the latitude and longitude of the geoReference to fit my map).

ebadi, 2020-12-10:
ok, thanks for explanation. I thought you came up with that number using a more complex operation. 
AFAIK geoReference tag specifies a transformation, you can [pass parameters](www.euref.eu/documentation/Tutorial2019/t-01-Evers.pdf) to the geo-reference you mentioned like this:
```
$ echo 1 2  | proj  +proj=tmerc +lat_0=37.35429341239328 +lon_0=-122.0859797650754 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs
7748802.31	15460124.93
```
According to [opendrive](https://releases.asam.net/OpenDRIVE/1.6.0/ASAM_OpenDRIVE_BS_V1-6-0.html#_georeferencing_in_opendrive) standard, this projection tag is not mandatory and not having it fine.

> There shall be no more than one definition of the projection. If the definition is missing, a local Cartesian coordinate system is assumed.

I also [found that](https://github.com/carla-simulator/carla/issues/1264) you can pass the projection value in two forms. If you want to try changing them, maybe you can try the simpler form as below:
```
<geoReference><![CDATA[+lat_0=4.9000000000000000e+1 +lon_0=8.0000000000000000e+0]]></geoReference>
```

yangyueren, 2020-12-21:
> ok, thanks for explanation. I thought you came up with that number using a more complex operation.
> AFAIK geoReference tag specifies a transformation, you can [pass [...]

Thanks for your reply. I will have a try.

Axel1092, 2020-12-22:
Hi @yangyueren, we currently do not export georeference in OpenDRIVE generated from OSM maps. The reason being we still need to implement and expose projection parameters in order to make it usable.

EchoYJJ, 2020-12-23:
@Axel1092 Hi, I met issues about map distortion about osm2odr.
I use the carla.Osm2Odr.convert and the main code lists below:
_f = open("D:\HHYtest\\map.osm", 'r',encoding='utf-8')
osm_data = f.read()
f.close()
settings = carla.Osm2OdrSettings()
xodr_data = carla.Osm2Odr.convert(osm_data, settings)
f = open("D:\HHYtest\\map.xodr", 'w',encoding='utf-8')
f.write(xodr_data)
f.close()_

After that, I import my xodr into Roadrunner. As you say, my xodr don't have any georeference or projection, so I "Set Custom Projection" in Roadrunner. I set the proper Lat&Long to "Use Tranverse Mercator At".
[image]
However, the distrotion still occurs. Do you know how to solve it? Thanks in advance.
[image]
[image]

thillRobot, 2021-01-23:
I am also experiencing a similar angular distortion issue when converting from **osm** to **xodr** in 0.9.11 build from source. This is also mentioned in #3009. When I convert the same file with `netconvert` from `sumo` the the result is not distorted, but the junctions must be fixed manually. Has anyone solved this problem?

JHMeusener, 2021-03-03:
Hello, if anyone is interested, i wrote a small workaround for the projection-issue with the Carla OSM Converter.
[Carla Projection Workaround for OSM to XODR Conversion](https://github.com/JHMeusener/CarlaSimpleXODRProjectionWorkaround)

closed 2021-03-08 by corkyw10

thillRobot, 2021-08-04:
I just tested the workaround from @JHMeusener with a map from the US near lon: 36.17, lat: -85.5, and the .osm and .xodr from CARLA matched well without distortion or changing the projection setting. I want to add that the modified .osm file did not show the map when loaded in Roadrunner, but the original .osm worked as expected. Thank you @JHMeusener for this contribution.

thillRobot, 2021-08-04:
I am also interested in the potential fix from @Axel1092, but this is not a priority for me at this time. I tried, but the dev branch did not compile on my machine. Maybe there is another way to apply the fix but I am not sure.