carla-simulator/carla #3688: Error when importing SUMO integration BridgeHelper

svanimisetti, 2020-12-09:
I am using CARLA 0.9.10. Not sure if this is fixed in latest release. When I run the SUMO "run_synchronization.py" example from its folder, the script works fine. However, I created another script at an alternate location. When try to import the SUMO integration package (C:/CARLA/Co-Simulation/Sumo/sumo_integration), I get the following error.

>File "C:/CARLA/Co-Simulation/Sumo/sumo_integration/bridge_helper.py", line 37, in BridgeHelper
>with open('data/vtypes.json') as f:
>FileNotFoundError: [Errno 2] No such file or directory: 'data/vtypes.json'

I modified the bridge_helper.py script with following change to make it work in a generic manner.

>Add Line 14: import os
>modify Line37-38:
>vtypes_file = os.path.join(os.path.dirname(os.path.realpath(__file__)),'../data/vtypes.json')
>with open(vtypes_file) as f:
>    _VTYPES = json.load(f)['carla_blueprints']

Is this the best way to modify the bridge helper so that when I pass the script to another user, he can run it without issues?

corkyw10, 2021-03-08:
@Axel1092 Could you follow up on this please?

joel-mb, 2021-03-16:
Hi @svanimisetti! Thanks for reporting that. I've opened a PR to include your changes.

closed 2021-03-16 by joel-mb

odajani, 2021-06-22:
I am using CARLA Nightly Build (Windows). I am following steps in SUMO co-simulation page https://carla.readthedocs.io/en/0.9.11/adv_sumo/  

cd ~/carla
./CarlaUE4.sh
cd PythonAPI/util
python3 config.py --map Town04
cd ~/carla/Co-Simulation/Sumo

everything is working fine but when I typed 
python3 run_synchronization.py examples/Town04.sumocfg  --sumo-gui  
I am getting the following error

File ".\run_synchronization.py", line 49, in <module>
    from sumo_integration.bridge_helper import BridgeHelper  # pylint: disable=wrong-import-position
  File "C:\Users\Documents\CARLA\WindowsNoEditor\Co-Simulation\Sumo\sumo_integration\bridge_helper.py", line 23, in <module>
    from .sumo_simulation import SumoSignalState, SumoVehSignal
  File "C:\Users\Documents\CARLA\WindowsNoEditor\Co-Simulation\Sumo\sumo_integration\sumo_simulation.py", line 25, in <module>
    import lxml.etree as ET  # pylint: disable=import-error
ModuleNotFoundError: No module named 'lxml'

I checked BridgeHelper file   and here is 23 to 50 lines 

from .sumo_simulation import SumoSignalState, SumoVehSignal
# =============================================================================
# -- Bridge helper (SUMO <=> CARLA) ----------------------------------------------------------------
# =============================================================================
class BridgeHelper(object):
    """
    BridgeHelper provides methos to ease the co-simulation between sumo and carla.
    """
    blueprint_library = []
    offset = (0, 0)
    _vtypes_path = os.path.join(os.path.dirname(os.path.realpath(__file__)), "..", "data",
                                "vtypes.json")
    with open(_vtypes_path) as f:
        _VTYPES = json.load(f)['carla_blueprints']
    @staticmethod
    def get_carla_transform(in_sumo_transform, extent):
        """
        Returns carla transform based on sumo transform.
        """
        offset = BridgeHelper.offset
        in_location = in_sumo_transform.location
        in_rotation = in_sumo_transform.rotation

Can you please advise how to fix this error?