carla-simulator/carla #3692: Significant slow down when there are other vehicles

dotchen, 2020-12-09:
Hello CARLA team!

I noticed that ever since version 0.9.7 where the traffic manager (TM) is introduced, the simulator siginificant slows down when there are other vehicles and pedestrians spawned.

For reference, I was benchmarking my agent in the NoCrash benchmark, where the empty scenario takes me 6 hour to finish, but the regular one, with 20 vehicles and 50 pedestrians, takes 42 hour (!!). The setup for both is exactly the same, except for the vehicles and pedestrians.

Is this due to the communication overhead between TM and the server, or I am running things incorrectly? Is there anyway to speed it up?

Thanks in advance!

corkyw10, 2021-03-08:
@jackbart94 would you be able to have a look at this please?

jackbart94, 2021-03-10:
Could you provide more information about your issue? Are you using CARLA 0.9.11 or another version? What are your PC specifics?

dotchen, 2021-03-10:
@jackbart94 The benchmark mentioned above was done in CARLA 0.9.9 but I've experienced this from CARLA 0.9.8 to CARLA 0.9.11. I am using Ubuntu 18.04 with 256G RAM.

jackbart94, 2021-03-25:
I wanted to know more about your graphic card, should've been more specific. That's the most important thing.

I don't know much about those datasets, nor how long are they supposed to last. A suggestion that I could give is to [read this post I've recently written on the forums about improving FPS](https://forum.carla.org/t/whats-the-best-way-to-simulate-in-scenes-with-a-lot-of-traffic/867/2) and see if it helps. It should reduce the number of hours.

dotchen, 2021-03-26:
Thanks a lot for the tip about using hybrid mode! I'll give it a try. Regarding hardware I've been using Titan X (pascal) and Titan Xp, as well as a GTX 1080Ti, and it slows down whenever there are more vehicles/peds spawned on the map, regardless hardware.

closed 2021-11-18 by jackbart94