carla-simulator/carla #3696: Tags in semantic segmentation goes to all "255"

trilokimodi, 2020-12-10:
I am using Carla 0.9.10. I am trying to extract semantic segmentation information. The document says 
"The server provides an image with the tag information encoded in the red channel. A pixel with a red value of x displays an object with tag x. The following tags are currently available".

My code - 

img = data.convert(ColorConverter.CityScapesPalette)
img = np.array(data.raw_data)
img = img.reshape((img_height, img_width, 4))

The unique pixel values in img are
(140, 150, 230, 255)
(156, 102, 102, 255)
(50, 120, 170, 255)
(70, 70, 70, 255)
(232, 35, 244, 255)
(35, 142, 107, 255)
(150, 60, 45, 255)
(128, 64, 128, 255)
(30, 170, 250, 255)
(100, 170, 145, 255)
(142, 0, 0, 255)
(50, 234, 157, 255)
(153, 153, 153, 255)
(81, 0, 81, 255)
(60, 20, 220, 255)
(160, 190, 110, 255)

Notice 255 in 4th column. I expected tags in the red channel as mentioned in Carla docs. Would be very nice if someone can give clarity on my issue. Thanks.

corkyw10, 2021-03-09:
@trilokimodi Thank you for your patience

@DSantosO are you able to have a look at this please?

winyouc, 2021-11-15:
hello,had the problem been solved?
I met the same situation now.
If you know how to solve it ,could you please tell me the solution?

winyouc, 2021-11-15:
I find the solution! You should choose the cc.Raw model( keyborad 4 camera, sensor.index =4 ),and then can get the tag between[0,22]

closed 2022-04-29 by stale[bot]