carla-simulator/carla #3698: Subdivision types of road line in segmentation camera with build-in carla map

jerryliu20001, 2020-12-11:
Hi all:
In my situation, I need to separate the rode line into single lines , double lines, zebra crossing, etc. for segmentation.
The problem is the road lines in CARLA map mostly seems to be combined together, looks like the yellow region selected in following picture:
[image]
Is it possible the separate the road lines, or it is just fixed when creating the map?
Currently, I just deleted the default road lines and drew the new ones, but it was extremely time-consuming ...
I thought it is related to the map creation.
Does anyone have any suggestion ?
Thanks in advance !!

corkyw10, 2021-03-09:
@XGodina could you follow up on this please?

XGodina, 2021-03-15:
Hi @jerryliu20001,
Yes, it is !! However, it's a long process. You shall all groups of lanes markings and export to a 3d software and after separately in any parts that you need. After that, you can import it again into the map. It's a surgical operation.

jerryliu20001, 2021-03-17:
Hi @XGodina ,
Thanks for your reply, but I have another question 
I can export the lanemark into .fbx file but I am not sure is it possible to separate the .fbx into parts
Do you have any suggestion for the CAD software which is able to split .fbx file?
Or maybe the exported file is not .fbx?

Thanks!!

closed 2023-06-02 by stale[bot]