carla-simulator/carla #3699: What's the reason for this “”? How to solTypeError: No registered converter was able to produce a C++ rvalue of type class carla::rpc::Command from this Python object of type NoneTypeve this problem?

KangChou, 2020-12-11:
for response in client.apply_batch_sync(batch):
TypeError: No registered converter was able to produce a C++ rvalue of type class carla::rpc::Command from this Python object of type NoneType

KangChou, 2020-12-14:
my code:

import glob
import os
import sys

# import future
# import pygame
# 添加系统变量
'''
try 
finally 函数体格式

def 函数名（）：
定义函数
'''
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla
import argparse
import logging
import random

def main():
    try:
        client = carla.Client('127.0.0.1', 2000)  # 默认地址
        print('create client')
        client.set_timeout(2.0)
        world = client.get_world()
        print('crated VR world')

        bp_library = world.get_blueprint_library()
        '''
        #获得blueprints 库-API 如下
        #carla.BlueprintLibrary
        #find(id)
        #filter(wildcard_pattern)
        #__getitem__(pos)
        #__len__()
        #__iter__()
        '''

        bps = bp_library.filter('vehicle.*')

        '''
        #获得了所有车的属性集合--matrix
        #进一步筛选
        '''
        bps = [x for x in bps if int(x.get_attribute('number_of_wheels')) == 4]

        '''获得所有的2轮车'''
        bps = [x for x in bps if not x.id.endswith('isetta')]
        bps = [x for x in bps if not x.id.endswith('carlacola')]

        '''获得可用生成点'''
        spoints = world.get_map().get_spawn_points()
        number_of_spawn_points = len(spoints)

        print(number_of_spawn_points)

        batch = []
        actor_list = []

        i = 0
        while i < 10:
            spoint = random.choice(spoints)
            bp = random.choice(bps)
            actorlist = world.spawn_actor(bp, spoint).set_autopilot(enabled=True)
            batch.append(actorlist)
            # set_autopilot-- 设置这些车辆自动行驶
            i = i + 1

        '''
        记录生成的actor 对象，
        提供摧毁他们的途径,下面的代码有问题，报错。TypeError: No registered converter was able to produce a C++ r
        '''
        ####
        for response in client.apply_batch_sync(batch):
            if response.error:
                logging.error(response.error)
            else:
                actor_list.append(response.actor_id)
        print('spawned %d vehicles, press Ctrl+C to exit.' % len(actor_list))
        #
        while True:
            world.wait_for_tick()
    #
    finally:
        #
        print('\ndestroying %d actors' % len(batch))
        client.apply_batch([carla.command.DestroyActor(x) for x in actor_list])

####

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')

corkyw10, 2021-03-09:
@doterop could you follow up on this please?

doterop, 2021-03-23:
Hi @KangChou 

You are generating the batch wrong. It should be filled with commands. Take a look at spawn_npc.py

closed 2021-03-23 by doterop