carla-simulator/carla #3701: Behaviour agent bug: incoming_waypoint becoming None

dHonerkamp, 2020-12-11:
Hi, I am using the latest version of the pythonAPI, built from source.

It seems that due to the waypoint buffering the behaviour_agent can run into the case where `self.incoming_waypoint` is None, leading to a failure when checking `self.incoming_waypoint.is_junction` in line 417 of carla/PythonAPI/carla/agents/navigation/behavior_agent.py

Probably this is a one-line fix, though I am not sure I understand the difference between the waypoint buffer and the waypoint queue enough to know if this just an unhandled case or if this should never be possible.

corkyw10, 2021-03-09:
@dHonerkamp thank you for your patience

@glopezdiest are you able to follow up on this please?

glopezdiest, 2021-03-10:
Yeah, we'll take a look

glopezdiest, 2021-07-21:
Ups, I completely forgot to report about this. Currently, that function is indeed badly created and returns none if the buffer is very small. We are improving the agents API and this will be solved when it comes out. As of right now, a check should be added after this function is used, to check if it returns None

closed 2021-07-21 by glopezdiest