carla-simulator/carla #3705: Roadmap for 4.25 or 4.26 Upgrade?

jimmyw404, 2020-12-11:
Hello CARLA team, thank you for your hard work on this simulator!

With 4.26 being recently released, are there any plans to upgrade to 4.25 or 4.26? Thank you!

https://www.unrealengine.com/en-US/blog/unreal-engine-4-26-released

qhaas, 2021-01-29:
Since my team needed to get headless vulkan working without X ( Issue #3800 ) for our HPC runs to avoid the OpenGL artifacts 
of Issue #3377 , and UE 4.25+ was needed for that... created a patch file that will get CARLA 0.9.11 working with UE 4.25.4 over in [Issue #3214](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155).

JohanVer, 2021-03-01:
@qhaas that is great, thank you! Do you have any idea if the changes that you provided let carla also compile on 4.26 (I need the new water system) ?

qhaas, 2021-03-01:
As I mentioned in [Issue 3214](https://github.com/carla-simulator/carla/issues/3214#issuecomment-773284624), Epic added a a `DEPRECATED` suffix to several API calls used by CARLA with UE 4.25.  I haven't checked, but those were likely removed in UE 4.26 and will require a heavier lift than going to 4.25.  The good news is that it appears CARLA will be migrating to UE 4.26 in the future, per [Issue 3865](https://github.com/carla-simulator/carla/issues/3865#issuecomment-777357195).

corkyw10, 2021-08-05:
Hey everyone, CARLA 0.9.12 on UE4.26 was released this week. Thanks for your patience!

closed 2021-08-05 by corkyw10