carla-simulator/carla #3707: carla client doesn't wait for server to start even with set_timeout

qhaas, 2020-12-11:
When I use [`set_timeout`](https://carla.readthedocs.io/en/0.9.10/python_api/#carlaclient) with the CARLA PythonAPI, I expect it to periodically check to see if the specified port (default 2000) is receiving connections and if so, connect to it before the timeout is over and proceed.  It doesn't appear to do so... if the server isn't up/running before the carla client is called, it appears to simply fail after the timeout is up... consider the following two scripts to launch a CARLA 0.9.10 server/client pair, with a slightly modified `tutorial.py`:

```
$ grep client.set_timeout PythonAPI/examples/tutorial.py 
        client.set_timeout(6.0)

$ cat testSetTimeout.sh
#!/bin/bash
./CarlaUE4.sh &
# start the client immediately
python3 PythonAPI/examples/tutorial.py

$ cat testSetTimeoutBlock.sh
#!/bin/bash
./CarlaUE4.sh &
# wait until port 2000 is listening
for ((jj=0; jj<6; jj++)); do
 if [ $(lsof -nP -iTCP:2000 -sTCP:LISTEN | wc -l) -gt 1 ]; then
  break
 fi
 sleep 1
done
python3 PythonAPI/examples/tutorial.py
```

The first one fails, despite giving plenty of time for the server to start.  After killing the server started by the first script, the 2nd script succeeds:
```
$ ./testSetTimeout.sh
...
RuntimeError: time-out of 6000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
...
```
```
$ ./testSetTimeoutBlock.sh
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
created vehicle.carlamotors.carlacola
created sensor.camera.depth
moved vehicle to Location(x=214.432999, y=9.808371, z=-0.005925)
created vehicle.nissan.micra
created vehicle.citroen.c3
created vehicle.kawasaki.ninja
created vehicle.volkswagen.t2
...
```

qhaas, 2021-01-04:
Seems to still occur in CARLA 0.9.11

qhaas, 2021-01-08:
This appears to be an issue with the underlying rpclib dependency and requires the entire client to be re-instantiated to proceed if the server isn't up/running/ready to receive connections when the client reaches out to it.

qhaas, 2021-01-10:
We implemented a workaround where we basically just try/catch the timeout in the CARLA side of the C++ code, re-initialize the rpc client, and retry a few times.  The patch file for CARLA 0.9.8 is below, but this is a stopgap since this inability to recover from a late starting server likely should be addressed at the rpclib level, so not going to create a merge request.

Speaking of rpclib, given [carla has forked rpclib](https://github.com/carla-simulator/rpclib/tree/v2.2.1_c4), likely due to [upstream abandonment](https://github.com/rpclib/rpclib/issues/229).  @bernatx it might be worth examining [another user's fork](https://github.com/qchateau/rpclib/commits/master) for fixes, but it doesn't look like this particular shortcoming has been addressed in that fork either (based on commit logs).

[carla_issue_3707.txt](https://github.com/carla-simulator/carla/files/5792874/carla_issue_3707.txt)

qhaas, 2021-05-24:
It would appear that the maintainer of the original rpclib is back [and looking for someone to takeover](https://github.com/rpclib/rpclib/issues/273).

sztomi, 2021-07-23:
@qhaas to be clear: there is a new maintainer (@qchateau) tending for the project (merging PRs and bugfixes mostly). I opened that issue because this maintenance doesn't cover new development - in case someone is interested in that, they would be welcome as another maintainer.

closed 2023-06-02 by stale[bot]