carla-simulator/carla #3708: OpenDRIVE Standalone Mode issues: importing my own xodr into CARLA by python config.py -x opendrive/test.xodr, but nothing appears except the background.

EchoYJJ, 2020-12-12:
I've been experiencing issues with OpenDRIVE standalone.
After following the tutorials at "https://carla.readthedocs.io/en/latest/adv_opendrive/", I successfully  tested with the new TownBig provided by CARLA by "python config.py -x opendrive/TownBig.xodr".
However, when I apply it for my own "test.xodr". nothing appears except the background and no error/warning shows up.

After getting an osm file, I obtained my own "test.xodr" via "carla.Osm2Odr" and "carla.Osm2OdrSettings" which you can find at "https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/#2-convert-to-opendrive-format".
Also, I wonder how to adjust xodr in CARLA. Is the XODR format 2D or 3D? Is using Roadrunner a must?
Looking forward to your reply, thanks!
[image]
[image]

corkyw10, 2021-03-09:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-15:
Hi @EchoYJJ, it is likely that your OpenDRIVE is not centered. The standalone mode will generate the road according the OpendDRIVE information. If the road is not centered you will need to move the camera to look for it. I suggest you try the `manual_control.py` to test if the road has been generated as it will spawn a car on a road.

closed 2021-03-24 by EchoYJJ

seuwcs, 2021-04-14:
Using Roadrunner is the best choice @EchoYJJ @

CaffreyXu, 2022-04-13:
Hello @Axel1092 , it seems that the lane lines are not shown in UE. How can i solve this problem? Thank you!

Daisuke7155, 2022-08-29:
Hello CARLA Team,
Now I have same issue and have not solved it yet. Could you tell me any solutions?

My condition
 -CARLA version: 0.9.11
 -Platform/OS: windows10

My issue
 - Nothing appears except the background and no error/warning shows up for my own .xodr file.
 - I successfully tested with TownBig provided by CARLA by "python config.py -x opendrive/TownBig.xodr".
 - My .xodr file can be opened by the other viewer. 
 - I already tried the manual_control.py to test on the world from my .xodr but roads didn't appear there and the car fell down.

I just started studying Carla and this is my first question on the forum. Thank you for your understanding.

CaffreyXu, 2022-08-29:
hi, @Daisuke7155. I also spent months on the problem. It seems that in this standalone mode, Carla just can only show the lane. You can see the solution here: https://github.com/carla-simulator/carla/pull/3990
If you look carefully in the code, following issue 3990, you will find that Carla didn't pass the lane information to UE.
However, it is not easy to show lanes in UE, which relates to some visualization concepts in UE, and I also don't know how to solve them.
Finally, I recommend you to try reading more issues on GitHub.
Wish you good luck~
https://github.com/carla-simulator/carla/issues/3161

seuwcs, 2022-08-29:
if you use carla for research, it is highly recommended that you build carla from source.on Ubuntu

Daisuke7155, 2022-08-29:
@CaffreyXu
Thank you for your reply. I understood this is a problem difficult to resolve soon. I'll read #3990 and #3161 after this, and other issues on Github more.

Now can I ask one more question? How do you think why we can read TownBig.xodr provided by CARLA as a .xodr sample?? 

@seuwcs 
Thank you for your comment. It seems carla built on ubuntu has less issues than windows. I'll arrange it soon(but I don't have Ubuntu environment now and I'm not familiar with that so I have to study that.)