carla-simulator/carla #3709: Question about spawning vehicles without Traffic Manager

yasser-h-khalil, 2020-12-12:
Hello,

Is it possible to spawn surrounding vehicles and set them to autopilot without traffic manager? I tried once, and the surrounding vehicles did not move.

Thanks

ll7, 2021-01-30:
I tried to spawn one vehicle with the autopilot and the vehicle didn't move. I wonder why.

This is my code and I can't find the error:

```python
import carla
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10.0) # seconds
# client.reload_world()

world = client.get_world()
world = client.load_world('Town03')

# traffic manager
tm = client.get_trafficmanager(8000)
tm.set_random_device_seed(0)
tm.global_percentage_speed_difference(-5.0)
# tm.global_distance_to_leading_vehicle(1)

blueprint_library = world.get_blueprint_library()

spectator = world.get_spectator()

vehicle_spawn_points = world.get_map().get_spawn_points()

tesla_bp = blueprint_library.filter('model3')[0]

tesla = world.try_spawn_actor(tesla_bp, vehicle_spawn_points[2])

spectator.set_transform(vehicle_spawn_points[2])

tm_port = tm.get_port()
tesla.set_autopilot(True, tm_port)
tm.auto_lane_change(tesla, True)
tm.distance_to_leading_vehicle(tesla, 1.0)
tm.vehicle_percentage_speed_difference(tesla, 0)

for i in range(10):
    time.sleep(1)
    snapshot = world.get_snapshot()
    print(snapshot.timestamp)

tesla.destroy()
```

ll7, 2021-01-30:
That is a reduced spawn_npc.py file that worked for me:

```python
#!/usr/bin/env python

import carla
import random
import logging
import time

def main():
    vehicles_list = []

    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0) # seconds

    synchronous_master = False

    try:
        world = client.get_world()

        traffic_manager = client.get_trafficmanager(8000)
        traffic_manager.set_global_distance_to_leading_vehicle(1)
        traffic_manager.set_random_device_seed(0)

        blueprints = world.get_blueprint_library().filter('model3')

        spawn_points = world.get_map().get_spawn_points()
        
        SpawnActor = carla.command.SpawnActor
        SetAutopilot = carla.command.SetAutopilot
        # SetVehicleLightState = carla.command.SetVehicleLightState
        FutureActor = carla.command.FutureActor

        # spawn vehicle

        blueprint = blueprints[0]
        transform = spawn_points[2]

        batch = []

        # TODO what is the driver_id used for?
        if blueprint.has_attribute('driver_id'):        
            driver_id = random.choice(blueprint.get_attribute('driver_id').recommended_values)
            blueprint.set_attribute('driver_id', driver_id)

        blueprint.set_attribute('role_name', 'autopilot')
        batch.append(SpawnActor(blueprint, transform)
            .then(SetAutopilot(FutureActor, True, traffic_manager.get_port())))

        for response in client.apply_batch_sync(batch, synchronous_master):
            if response.error:
                logging.error(response.error)
            else:
                vehicles_list.append(response.actor_id)

        spectator = world.get_spectator()
        spectator.set_transform(transform)

        time.sleep(20)
    finally:
        print('\ndestroying %d vehicles' % len(vehicles_list))
        client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])

        time.sleep(0.5)

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```

jackbart94, 2021-03-10:
> Hello,
> 
> Is it possible to spawn surrounding vehicles and set them to autopilot without traffic manager? I tried once, and the surrounding vehicles did not move.
> 
> Thanks

No, you can't.

> That is a reduced spawn_npc.py file that worked for me:
>
> ```python
> #!/usr/bin/env python
>
> import carla
> import random
> import logging
> import time
>
> def main():
>     vehicles_list = []
>
>     client = [...]

It works because now you are spawning vehicles correctly.

Closing.

closed 2021-03-10 by jackbart94

esq0604, 2022-09-14:
Hello whathello .
I have a question .

I saw that autopilot is registered in tm and moves.

Is dirve_id an id to register with tm?