carla-simulator/carla #3710: Vehicle slowing down after set_target_velocity

bilalhejase, 2020-12-12:
Hi everyone, when I try to set the initial velocity of my vehicle, using ```set_target_velocity()```, to a value such as 25m/s, the speed of the vehicle quickly begins to drop till around 18m/s before accelerating again. I am applying a throttle command constantly of 0.864 and no brake or steer. I also tick the world twice so that the command takes effect but the phenomenon still occured.

This is my code snippet for setting the initial velocity:
```
forward_vec = vehicle.get_transform().get_forward_vector()
velocity_vec = self.start_speed * forward_vec
vehicle.set_target_velocity(velocity_vec) 
```

How can I prevent my vehicle from dropping speed when setting a high initial velocity?

corkyw10, 2021-03-09:
@jackbart94 Could you follow up on this please?

jackbart94, 2021-03-10:
You need to change the PID values accordingly, whether you are using Traffic Manager or the agents. It's clear that it overshoots and tries to compensate, but the current values aren't fit for the velocity you want to reach. If you want more help, and think that PID isn't the issue, please provide a complete code snippet that we can run and debug.

bilalhejase, 2021-07-21:
The issue was that you need to also specify the appropriate gear in the VehicleControl. By default the gear is set to 0, which causes the car to initially deccelerate.

closed 2021-11-18 by jackbart94

XiaoLiSean, 2024-03-04:
Hi, everyone. I recently encountered a similar problem. I know it might sound a little bit weird, however, my development requires a pipeline as follows: set the vehicle speed to zero -> complete a certain task -> accelerate the vehicle speed to a desired value. Please see the code attached.

`
import argparse, sys
import carla

if __name__ == '__main__':

    # Get the host and port id
    argparser = argparse.ArgumentParser(description=__doc__)
    argparser.add_argument('--host', metavar='H', default='127.0.0.1',
        help='IP of the host server (default: 127.0.0.1)')
    argparser.add_argument('-p', '--port', metavar='P', default=2000, type=int,
        help='TCP port to listen to (default: 2000)')
    argparser.add_argument('--tm-port', metavar='P', default=8000, type=int,
        help='Port to communicate with TM (default: 8000)')
    argparser.add_argument('--record', action='store_true', help='Record video')
    args = argparser.parse_args()

    try:
        dt = 0.01
        # Connect to the client
        client = carla.Client(args.host, args.port)
        client.set_timeout(10.0)
        client.load_world('Town06')

        # Retrieve the world object
        world = client.get_world()
        settings = world.get_settings()
        settings.synchronous_mode = True
        settings.fixed_delta_seconds = dt
        world.apply_settings(settings)

        # Retrieve the spectator object and change the view point and angle
        spectator = world.get_spectator()
        transform = spectator.get_transform()
        transform.location = carla.Location(x=250, y=260, z=25.50) # left half of the view
        transform.rotation = carla.Rotation(roll=0, pitch=-60, yaw=-90)
        spectator.set_transform(transform)

        # Intialize the traffic manager
        traffic_manager = client.get_trafficmanager(args.tm_port)
        traffic_manager.set_synchronous_mode(True)
        world.tick() # this is neccesary to update the actor info
        
        # Initialize ego vehicle
        blueprint = world.get_blueprint_library().find('vehicle.lincoln.mkz_2020')
        blueprint.set_attribute('role_name', 'hero')
        transform = carla.Transform()
        transform.rotation = carla.Rotation(roll=0, pitch=0, yaw=0)
        transform.location = carla.Location(x=250, y=244.55, z=0.024)
        car = world.spawn_actor(blueprint, transform)
        world.tick() # this is neccesary to update the actor info

        # Accelerate to set speed
        v_k = 40 # in [m/s]
        a_k = 1 # in [m/s^2]
        car.set_target_velocity(carla.Vector3D(x=v_k)) # set_target_velocity in [m/s]
        car.enable_constant_velocity(carla.Vector3D(x=v_k)) # enforce set speed
        while True:
            car.set_target_velocity(carla.Vector3D(x=0.0)) # set_target_velocity in [m/s]
            car.enable_constant_velocity(carla.Vector3D(x=0.0)) # enforce set speed
            world.tick()

            # ---------------------------------------
            # some other code blocks to be execute after setting vehicle speed to zero
            # ---------------------------------------
            # ......................................
            # ---------------------------------------

            # accelerate
            v_k += a_k*dt
            car.set_target_velocity(carla.Vector3D(x=v_k)) # set_target_velocity in [m/s]
            car.enable_constant_velocity(carla.Vector3D(x=v_k)) # enforce set speed

            # settings = world.get_settings()
            # settings.fixed_delta_seconds = dt
            # world.apply_settings(settings)
            world.tick()
            # acceleration not working at larger speed
            print('Desired / Actual: {}/{}'.format(v_k, car.get_velocity().x))
            v_k = car.get_velocity().x

    except KeyboardInterrupt:
        print('Cancelled by user...')

    except Exception as err:
        print('Error: {}'.format(err))

    # Terminate carla and disconnet the port
    finally:
        pass
`
However, this piece of code cannot set the velocity (which remains zero instead) to the desired ones with world.tick() after a few loops. When I tried to uncomment the following lines, I somehow could get non-zero velocity, but the velocity is decreasing while I anticipate it to increase.
`            
# settings = world.get_settings()
# settings.fixed_delta_seconds = dt
# world.apply_settings(settings)
`
I'm personally not very familiar with the underlying Carla development, and explanations of these observations from experts will be much appreciated. Meanwhile,  I would be very grateful if I could possibly be provided with solutions to these issues or guidance for realizing the aforementioned pipeline (i.e., set the vehicle speed to zero -> complete a certain task -> accelerate the vehicle speed to a desired value).