carla-simulator/carla #3715: OpenScenario Double Lane Changer - Error message

nithaventhan1506, 2020-12-14:
Hey together,

i tried to use scenariorunner 0.9.10 for Carla 0.9.10 to start the OpenScenario DoubleLaneChanger. 
python scenario_runner.py --openscenario C:\..\DoubleLaneChanger_Carla.xosc

But i get this error message : 
WARNING:[SR:OpenScenarioConfiguration]: The C:\Tools\scenario_runner-0.9.10\srunner\examples/Catalogs/VehicleCatalogs/VehicleCatalog.xosc path for the Vehicle Catalog is invalid
WARNING:[SR:OpenScenarioConfiguration]: The C:\Tools\scenario_runner-0.9.10\srunner\examples/Catalogs/ControllerCatalogs/ControllerCatalog.xosc path for the Controller Catalog is invalid
WARNING:[SR:OpenScenarioConfiguration]: Wrong OpenDRIVE map in use. Forcing reload of CARLA world
WARNING: cannot parse georeference: ''. Using default values.
WARNING: In road 156 lane 4 no "<width>" parameter found under "<lane>" tag. Using default values.
WARNING: In road 153 lane 4 no "<width>" parameter found under "<lane>" tag. Using default values.
Traceback (most recent call last):
  File "scenario_runner.py", line 593, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 583, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 488, in run
    result = self._run_openscenario()
  File "scenario_runner.py", line 476, in _run_openscenario
    config = OpenScenarioConfiguration(self._args.openscenario, self.client)
  File "C:\Tools\scenario_runner-0.9.10\srunner\scenarioconfigs\openscenario_configuration.py", line 59, in __init__
    self._parse_openscenario_configuration()
  File "C:\Tools\scenario_runner-0.9.10\srunner\scenarioconfigs\openscenario_configuration.py", line 91, in _parse_openscenario_configuration
    self._set_actor_information()
  File "C:\Tools\scenario_runner-0.9.10\srunner\scenarioconfigs\openscenario_configuration.py", line 215, in _set_actor_information
    entry = OpenScenarioParser.get_catalog_entry(self.catalogs, catalog_reference)
  File "C:\Tools\scenario_runner-0.9.10\srunner\tools\openscenario_parser.py", line 211, in get_catalog_entry
    entry = catalogs[catalog_reference.attrib.get("catalogName")][catalog_reference.attrib.get("entryName")]
KeyError: 'VehicleCatalog'

Someone knows this Issue?

thanks!

nithaventhan1506, 2020-12-15:
Missing Catalog ... i got it

closed 2021-03-09 by corkyw10

yiyizi112, 2022-04-03:
I have the same problem, can you tell me how you solved it?