carla-simulator/carla #3731: Will future carla use pybind11 as Python API?

zeyu-hello, 2020-12-17:
Hi, 

Carla is a great work!
Carla uses `boost.python` as python api, and there is another python-c++ api `pybind11`. As far as I know, pybind11 is more pure and simple, while `boost.python` needs whole `boost lib` to be installed. So I want to know will carla use pybind11 as your python api inthe future?

Best regards!

alanhuang1990, 2023-05-10:
Using pybind11 to refactor python-c++ api also can provide typing information using  [mypy](https://mypy.readthedocs.io/en/stable/stubgen.html). This would make carla more self documenting.  It is realy an important feature for user that using modern IDEs like VSCode, Pycharm.

MarcelPiNacy-CVC, 2024-05-10:
Hi,

When I have some time I may look into whether pybind would degrade performance.
Are you asking because you want to have type info or because compiling boost is too slow?
If it's the second the current build system rework should significantly lessen this problem.

Marcel

qhaas, 2025-01-20:
This might resolve Issue #6547 as well.