carla-simulator/carla #3733: How to add attribute in sensor blueprint?

zeyu-hello, 2020-12-17:
### Description
Hi, I add one 'attribute' in [PythonAPI/carla/source/libcarla/SensorData.cpp](https://github.com/carla-
simulator/carla/blob/master/PythonAPI/carla/source/libcarla/SensorData.cpp#L275-L307), since I want to add a param to pass `lidar_type` and then use `make rebuild` to re-compile carla.
```c++
  class_<csd::LidarMeasurement, bases<cs::SensorData>, boost::noncopyable, boost::shared_ptr<csd::LidarMeasurement>>("LidarMeasurement", no_init)
    .add_property("horizontal_angle", &csd::LidarMeasurement::GetHorizontalAngle)
    .add_property("channels", &csd::LidarMeasurement::GetChannelCount)
    .add_property("lidar_type", &csd::LidarMeasurement::GetLidarType)
    .add_property("raw_data", &GetRawDataAsBuffer<csd::LidarMeasurement>)
    .def("get_point_count", &csd::LidarMeasurement::GetPointCount, (arg("channel")))
    .def("save_to_disk", &SavePointCloudToDisk<csd::LidarMeasurement>, (arg("path")))
    .def("__len__", &csd::LidarMeasurement::size)
    .def("__iter__", iterator<csd::LidarMeasurement>())
    .def("__getitem__", +[](const csd::LidarMeasurement &self, size_t pos) -> csd::LidarDetection {
      return self.at(pos);
    })
    .def("__setitem__", +[](csd::LidarMeasurement &self, size_t pos, const csd::LidarDetection &detection) {
      self.at(pos) = detection;
    })
    .def(self_ns::str(self_ns::self))
  ;

  class_<csd::SemanticLidarMeasurement, bases<cs::SensorData>, boost::noncopyable, boost::shared_ptr<csd::SemanticLidarMeasurement>>("SemanticLidarMeasurement", no_init)
    .add_property("horizontal_angle", &csd::SemanticLidarMeasurement::GetHorizontalAngle)
    .add_property("channels", &csd::SemanticLidarMeasurement::GetChannelCount)
    .add_property("lidar_type", &csd::SemanticLidarMeasurement::GetLidarType)
    .add_property("raw_data", &GetRawDataAsBuffer<csd::SemanticLidarMeasurement>)
    .def("get_point_count", &csd::SemanticLidarMeasurement::GetPointCount, (arg("channel")))
    .def("save_to_disk", &SavePointCloudToDisk<csd::SemanticLidarMeasurement>, (arg("path")))
    .def("__len__", &csd::SemanticLidarMeasurement::size)
    .def("__iter__", iterator<csd::SemanticLidarMeasurement>())
    .def("__getitem__", +[](const csd::SemanticLidarMeasurement &self, size_t pos) -> csd::SemanticLidarDetection {
      return self.at(pos);
    })
    .def("__setitem__", +[](csd::SemanticLidarMeasurement &self, size_t pos, const csd::SemanticLidarDetection &detection) {
      self.at(pos) = detection;
    })
    .def(self_ns::str(self_ns::self))
  ;
```
and I also add a line in [PythonAPI/examples/open3d_lidar.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/open3d_lidar.py#L136) as follows:
```python
def generate_lidar_bp(arg, world, blueprint_library, delta):
    """Generates a CARLA blueprint based on the script parameters"""
    if arg.semantic:
        lidar_bp = world.get_blueprint_library().find('sensor.lidar.ray_cast_semantic')
    else:
        lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')
        if arg.no_noise:
            lidar_bp.set_attribute('dropoff_general_rate', '0.0')
            lidar_bp.set_attribute('dropoff_intensity_limit', '1.0')
            lidar_bp.set_attribute('dropoff_zero_intensity', '0.0')
        else:
            lidar_bp.set_attribute('noise_stddev', '0.00001')

    lidar_bp.set_attribute('upper_fov', str(arg.upper_fov))
    lidar_bp.set_attribute('lower_fov', str(arg.lower_fov))
    lidar_bp.set_attribute('channels', str(arg.channels))
    lidar_bp.set_attribute('range', str(arg.range))
    lidar_bp.set_attribute('rotation_frequency', str(1.0 / delta))
    lidar_bp.set_attribute('points_per_second', str(arg.points_per_second))
    lidar_bp.set_attribute('lidar_type', str(arg.lidar_type))                                   # <<<<<<<<<<<<<<<<<  HERE!!!
    return lidar_bp
```
but when I ran it, it returns this:
```shell
IndexError: attribute 'lidar_type' not found
```
### Enviorment
python: 3.7.6
system: ubuntu 16.04

MisterioRemo, 2021-01-16:
Hello! Do you still need help?

I'm assuming you don't have your own lidar class and want to change RayCastSemanticLidar and RayCastLidar.
Let's say lidar_type variable has uint type so it can be cast to enum if needed.
Add lidar_type  to `Carla/Sensor/LidarDescription.h`.
```cpp
USTRUCT()
struct CARLA_API FLidarDescription
{
  GENERATED_BODY()

  // carla's code

  UPROPERTY(EditAnywhere)
  uint32 LidarType = 0u;
};
```

Then in `Carla/Actor/ActorBlueprintFunctionLibrary.cpp` modify `MakeLidarDefinition` and `SetLidar` functions.
```cpp
void UActorBlueprintFunctionLibrary::MakeLidarDefinition(
    const FString &Id,
    bool &Success,
    FActorDefinition &Definition)
{
  // carla's code
  FActorVariation LidarType;
  LidarType.Id = TEXT("lidar_type");
  LidarType.Type = EActorAttributeType::Int;
  LidarType.RecommendedValues = { TEXT("0") };

  // append LidarType to Definition.Variations for both ray_cast and ray_cast_semantic
  if (Id == "ray_cast") {
    Definition.Variations.Append(
        {Channels, Range, PointsPerSecond, Frequency, UpperFOV, LowerFOV,
            AtmospAttenRate, DropOffGenRate, DropOffIntensityLimit,
            DropOffAtZeroIntensity, StdDevLidar, LidarType});
  }
  else if (Id == "ray_cast_semantic") {
    Definition.Variations.Append(
        {Channels, Range, PointsPerSecond, Frequency, UpperFOV, LowerFOV, LidarType});
  }
  else {
    DEBUG_ASSERT(false);
  }

  Success = CheckActorDefinition(Definition);
}

// more code

void UActorBlueprintFunctionLibrary::SetLidar(
    const FActorDescription &Description,
    FLidarDescription &Lidar)
{
  // carla's code
  Lidar.LidarType=
      RetrieveActorAttributeToInt("lidar_type", Description.Variations, Lidar.LidarType);
}
```

Then rebuild carla.
Alternatively, you can modify `FActorDefinition GetSensorDefinition()` and `void Set(const FActorDescription &Description)` as described in [How to add a new sensor](https://carla.readthedocs.io/en/latest/tuto_D_create_sensor/) tutorial.

NurbekHalikulov, 2021-02-04:
Hey @MisterioRemo ,

many thanks for your answer. Actually I used your code as a reference.

However, I still have an issue. 

My task was to add these attributes
    sensor.set_attribute('dropoff_general_rate', str(0.2))
    sensor.set_attribute('dropoff_intensity_limit', str(0.75))
    sensor.set_attribute('dropoff_zero_intensity', str(0.25))
    sensor.set_attribute('noise_stddev', str(0.2))

to my sensor.lidar.ray_cast sensor.

As you said, I modiefied Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor  - ActorBlueprintFunctionLibrary.cpp with such attributes. Also I gave corresponding default values and it looks like now:

......
  // Horizontal FOV.
  FActorVariation HorizontalFOV;
  HorizontalFOV.Id = TEXT("horizontal_fov");
  HorizontalFOV.Type = EActorAttributeType::Float;
  HorizontalFOV.RecommendedValues = { TEXT("360.0") };
  // Atmospheric Attenuation Rate.
  FActorVariation AtmospAttenRate;
  AtmospAttenRate.Id = TEXT("atmosphere_attenuation_rate");
  AtmospAttenRate.Type = EActorAttributeType::Float;
  AtmospAttenRate.RecommendedValues = { TEXT("0.004") };
  // Noise seed
  FActorVariation NoiseSeed;
  NoiseSeed.Id = TEXT("noise_seed");
  NoiseSeed.Type = EActorAttributeType::Int;
  NoiseSeed.RecommendedValues = { TEXT("0") };
  NoiseSeed.bRestrictToRecommended = false;
  // Dropoff General Rate
  FActorVariation DropOffGenRate;
  DropOffGenRate.Id = TEXT("dropoff_general_rate");
  DropOffGenRate.Type = EActorAttributeType::Float;
  DropOffGenRate.RecommendedValues = { TEXT("0.45") };
  // Dropoff intensity limit.
  FActorVariation DropOffIntensityLimit;
  DropOffIntensityLimit.Id = TEXT("dropoff_intensity_limit");
  DropOffIntensityLimit.Type = EActorAttributeType::Float;
  DropOffIntensityLimit.RecommendedValues = { TEXT("0.8") };
  // Dropoff at zero intensity.
  FActorVariation DropOffAtZeroIntensity;
  DropOffAtZeroIntensity.Id = TEXT("dropoff_zero_intensity");
  DropOffAtZeroIntensity.Type = EActorAttributeType::Float;
  DropOffAtZeroIntensity.RecommendedValues = { TEXT("0.4") };
  // Noise in lidar cloud points.
  FActorVariation StdDevLidar;
  StdDevLidar.Id = TEXT("noise_stddev");
  StdDevLidar.Type = EActorAttributeType::Float;
  StdDevLidar.RecommendedValues = { TEXT("0.0") };

  if (Id == "ray_cast") {
	  Definition.Variations.Append({
		Channels,
		Range,
		PointsPerSecond,
		Frequency,
		UpperFOV,
		LowerFOV,
		AtmospAttenRate,
		NoiseSeed,
		DropOffGenRate,
		DropOffIntensityLimit,
		DropOffAtZeroIntensity,
		StdDevLidar,
		HorizontalFOV });
  }
  else if (Id == "ray_cast_semantic") {
	  Definition.Variations.Append({
		Channels,
		Range,
		PointsPerSecond,
		Frequency,
		UpperFOV,
		LowerFOV,
		HorizontalFOV });
  }
  else {
	  DEBUG_ASSERT(false);
  }
........

Afterwards, I made "make rebuild" of carla. When I run my script, I am still getting **IndexError: attribute 'dropoff_general_rate' not found**

Could you please tell me, what could be the reason? Or should I change content of other files also?

closed 2023-06-02 by stale[bot]