carla-simulator/carla #3735: semi-truck availability in carla

SeasonedLeo, 2020-12-17:
Hello 

I know this question was asked last year. I just wanted to check back and see if there is availability/capability of simulating semi-trucks in Carla? Last year it was mentioned that currently there is no capability but there are plans to add in the future. 
please advise.

Thanks

BAROBA-3d, 2021-01-02:
Hi SeasonedLeo,
I want this to happen too. The good news is that you can make a lone truck with making either a 4 wheeled version of it or by lifting one axle of the ground and drive around as a 4 wheeled version. 
The thing you want ( and I too) is to have a truck and trailer... I think that is a little bit harder to make... I am still working on ideas on how to do it, but I think that a 2 wheeled vehicle could make a good base for a fake truck-trailer ( frontwheel= truck, rear-wheel= trailer) Maybe one of the programmers can work with that :)

XGodina, 2021-01-11:
[Hi, @SeasonedLeo, @BAROBA-3d 
I think, I've talked with BAROBA, that at this moment we are working to improve our vehicle gallery. But we have in mind to implement semi-truck. In the next release, we will implement trucks without a trailer, although it will be possible to put one. In that moment we will start with that, I know it's a little time, but we want cars to look good

[image]

manonechu, 2021-02-06:
@XGodina
This is a good news! When will be the date of the next release? Thanks a lot.

closed 2021-07-21 by stale[bot]

frankeng, 2022-05-10:
Hi I working on a semitrailer truck.
So far I got the concept working. I added the trailer as an separate  4 wheeled vehicle in Carla. Added a towing vehicle with a hitch. and connected the the towing vehicle an trailer with the following blue print:
[image]
You can see the result here https://youtu.be/nr4RuLNotFo

Julbak, 2022-11-29:
Hi there, does the vehicle always gets attached at the middle? Wonder if it can be specified to attach at the rear so that when replacing the truck with another vehicle it looks correct.

frankeng, 2022-11-30:
I do not under stand your question, you want to connect a for example  car to the trailer?

GueneyOez, 2025-09-06:
@frankeng, do you still have the files for that truck + trailer? I used your blueprint, but the two vehicle never connect. The Blueprint (plus box and physics constraint) is on the trailers event graph, and I have a Hitch Socket on the Trailer and one on the Truck, but it neveer gets connected.