carla-simulator/carla #3740: 'PhysicsEngine/BodySetupEnums.h' file not found

MS012086, 2020-12-18:


Axel1092, 2020-12-21:
Hi @MS012086, you are compiling CARLA with unreal 4.26 but only 4.24 is supported in version 0.9.10.1. You need to switch to the supported version to be able to compile the project.

MS012086, 2020-12-22:
Dear Axel1092,

Thank you very much for your response.

I have tried with Unreal engine 4.24 software, following error message is received while launch the Carla. Can you please help me out regarding this issue.

Carla version 0.9.10.1
Unreal Engine 4.24
Ubuntu 18.04
Python 2.7

/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1244:8: error: class member cannot be redeclared
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1237:8: note: previous declaration is here
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
In file included from /home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanIndexBuffer.cpp:7:
In file included from Runtime/VulkanRHI/Private/VulkanRHIPrivate.h:41:
In file included from Runtime/VulkanRHI/Private/VulkanRHIPrivate.h:41:
In file included from Runtime/VulkanRHI/Private/VulkanResources.h:15:
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1244:8: error: class member cannot be redeclared
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1237:8: note: previous declaration is here
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1245:8: error: class member cannot be redeclared
In file included from Runtime/VulkanRHI/Private/VulkanResources.h:15:
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1244:8: error: class member cannot be redeclared
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1237:8: note: previous declaration is here
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1245:8: error: class member cannot be redeclared
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1245:8: error: class member cannot be redeclared
                void EnqueueBufferSuballocation(TRefCountPtr<VulkanRHI::FBufferSuballocation> Allocation);
                     ^
                void EnqueueBufferSuballocation(TRefCountPtr<VulkanRHI::FBufferSuballocation> Allocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1238:8: note: previous declaration is here
                void EnqueueBufferSuballocation(TRefCountPtr<VulkanRHI::FBufferSuballocation> Allocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1246:8: error: class member cannot be redeclared
                void EnqueueBufferSuballocationDirect(FBufferSuballocation* SubAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1239:8: note: previous declaration is here
                void EnqueueBufferSuballocationDirect(FBufferSuballocation* SubAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1238:8: note: previous declaration is here
                void EnqueueBufferSuballocation(TRefCountPtr<VulkanRHI::FBufferSuballocation> Allocation);
                     ^
In file included from /home/Test/UnrealEngine_4.24/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/VulkanRHI/Module.VulkanRHI.2_of_3.cpp:2:
                void EnqueueBufferSuballocation(TRefCountPtr<VulkanRHI::FBufferSuballocation> Allocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1238:8: note: previous declaration is here
In file included from /home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanTexture.cpp:7:
In file included from Runtime/VulkanRHI/Private/VulkanRHIPrivate.h:41:
In file included from Runtime/VulkanRHI/Private/VulkanResources.h:15:
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1244:8: error: class member cannot be redeclared
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1237:8: note: previous declaration is here
                void EnqueueResourceAllocation(TRefCountPtr<VulkanRHI::FOldResourceAllocation> ResourceAllocation);
                     ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1245:8: error: class member cannot be redeclared
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:1246:8: error: class member cannot be redeclared
                void EnqueueBufferSuballocationDirect(FBufferSuballocation* SubAllocation);
[... cut ...]
                                     ^
8 errors generated.
11 errors generated.
8 errors generated.
8 errors generated.
8 errors generated.
8 errors generated.
8 errors generated.
8 errors generated.
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/Test/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2
Test@Test-D-12680:~/carla$

Thanks in advance!
Thanks & Regards,
M Saravanan

Axel1092, 2020-12-22:
It seems like UE4 is missing some files, did the `./Setup.sh && ./GenerateProjectFiles.sh && make` command produce any error when compiling the engine?

MS012086, 2020-12-23:
Dear Axel1092,

Yes, I received following error message.

Test@Test-D-12680:~/UnrealEngine_4.24$ ./Setup.sh && ./GenerateProjectFiles.sh && make
Registering git hooks... (this will override existing ones!)
Fixing inconsistent case in filenames.
Setting up Mono
Checking dependencies...
Result: 0
Register the engine installation...
Installing a bundled clang toolchain
Toolchain already installed skipping.

Target arch set to: x86_64-unknown-linux-gnu
Building ThirdParty libraries

If you don't see SUCCESS message in the end, then building did not finish properly.
In that case, take a look into /home/Test/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/BuildThirdParty.log for details.

No third party libs needed to be built locally

**********  SUCCESS ****************
Setup successful.
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Writing project files... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Generating data for project indexing... 100%
Writing project files... 100%
bash "/home/Test/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
Fixing inconsistent case in filenames.
Setting up Mono
Building CrashReportClient...
Using 'git status' to determine working set for adaptive non-unity build (/home/Test/UnrealEngine_4.24).
Target is up to date
Total execution time: 0.82 seconds
bash "/home/Test/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClientEditor Linux Shipping  
Fixing inconsistent case in filenames.
Setting up Mono
Building CrashReportClientEditor...
Using 'git status' to determine working set for adaptive non-unity build (/home/Test/UnrealEngine_4.24).
Target is up to date
Total execution time: 0.74 seconds
bash "/home/Test/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/Build.sh" ShaderCompileWorker Linux Development  
Fixing inconsistent case in filenames.
Setting up Mono
Building ShaderCompileWorker...
Using 'git status' to determine working set for adaptive non-unity build (/home/Test/UnrealEngine_4.24).
Creating makefile for ShaderCompileWorker (no existing makefile)
------- Build details --------
Using toolchain located at '/home/Test/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/Test/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '8.0.1' (string), 8 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/Test/UnrealEngine_4.24/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Target is up to date
Total execution time: 1.02 seconds
[... 551 lines cut ...]
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:420:57: error: class member cannot be redeclared
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:394:57: note: previous definition is here
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:424:57: error: class member cannot be redeclared
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:394:57: note: previous definition is here
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:428:57: error: class member cannot be redeclared
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
/home/Test/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanResources.h:394:57: note: previous definition is here
        inline TRefCountPtr<VulkanRHI::FOldResourceAllocation> GetResourceAllocation()
                                                               ^
8 errors generated.
Makefile:908: recipe for target 'UE4Editor' failed
make: *** [UE4Editor] Error 5
Test@Test-D-12680:~/UnrealEngine_4.24$ 

Thanks & Regards,
Saravanan

corkyw10, 2021-03-09:
@MS012086 are you still facing this issue?

Axel1092, 2021-03-15:
@MS012086, sorry for the late reply. I have never seen these errors when compiling unreal. Make sure you follow the instructions in the documentation (https://carla.readthedocs.io/en/latest/build_linux/), also check the installation guide in the unreal documentation itself: https://docs.unrealengine.com/en-US/SharingAndReleasing/Linux/BeginnerLinuxDeveloper/SettingUpAnUnrealWorkflow/index.html

MS012086, 2021-03-18:
Hi @Axel1092, Thank you so much for your reply. I have resolved the issue. While adding a new vehicle into carla, i have missed one step due to that reason the above error is occurred. Now it is working fine.

closed 2021-03-18 by corkyw10