carla-simulator/carla #3743: make PythonAPI & make launch fails for Windows 10: probably C complier problems

EchoYJJ, 2020-12-19:
My setting:
Carla: git-clone from 'https://github.com/carla-simulator/carla'
cmake: 3.19.1
make: GNU Make 3.81
Windows 10 (no ubuntu)
VS: 2017 
I mostly follow the instructions here "https://carla.readthedocs.io/en/latest/build_windows/". It requires two additional elements.
_Windows 8.1 SDK. & x64 Visual C++ Toolset._ When I search "x64 Visual C++ Toolset" I install the" VC++ 2017 version 15.8 v14.15 toolset" Also, I already have the version 15.9.
[image]

Then, I renew my environment variables like below.
[image]

However, when I move to "carla" folder, and run "make Python" or "make launch", it shows the warning below. I'm not certain about whether my "cl.exe" is right. Also, I wonder if Ubuntu is easier to bulid.
If you have some insights, please give me some advices, thanks!

PS D:\carla-master> make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "D:\carla-master\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla-master\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "D:\carla-master\Build\zlib-source\".
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.15.26731.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: D:/VS2017/VC/Tools/MSVC/14.15.26726/bin/HostX64/x64/cl.exe
-- Check for working C compiler: D:/VS2017/VC/Tools/MSVC/14.15.26726/bin/HostX64/x64/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    **"D:/VS2017/VC/Tools/MSVC/14.15.26726/bin/HostX64/x64/cl.exe"

  is not able to compile a simple test program.**

  It fails with the following output:

    Change Dir: D:/carla-master/Build/zlib-source/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake /nologo cmTC_e9eca\fast &&       D:\VS2017\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\nmake.exe  -f CMakeFiles\cmTC_e9eca.dir\build.make /nologo -L                  CMakeFiles\cmTC_e9eca.dir\build
    Building C object CMakeFiles/cmTC_e9eca.dir/testCCompiler.c.obj
        D:\VS2017\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\cl.exe @C:\Users\EchoYJJ\AppData\Local\Temp\nmC18E.tmp
    testCCompiler.c
    Linking C executable cmTC_e9eca.exe
        "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_e9eca.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests -- D:\VS2017\VC\Tools\MSVC\14.15.26726\bin\HostX64\x64\link.exe /nologo @CMakeFiles\cmTC_e9eca.dir\objects1.rsp @C:\Users\EchoYJJ\AppData\Local\Temp\nmC1FD.tmp
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_e9eca.dir/manifest.res CMakeFiles\cmTC_e9eca.dir/manifest.rc" failed (exit code 0) with the following output:
    The system could not find the specified file NMAKE : fatal error U1077: “"C:\Program Files\CMake\bin\cmake.exe"”:  return code“0xffffffff”
    Stop.
    NMAKE : fatal error U1077: “D:\VS2017\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\nmake.exe”: return code“0x2”
    Stop.
  **CMake will not be able to correctly generate this project.**
 Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "D:/carla-master/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "D:/carla-master/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

EchoYJJ, 2020-12-19:
I noticed that my UE version is 4.26, and if I run "make clean", it shows the warning:

_PS D:\carla-master> make clean
-[Package]: [Batch params]: --clean --ue-version 4.24
fatal: not a git repository (or any of the parent directories): .git

-[Package]: [ERROR] Carla Version is not set
make: *** [clean] 错误 1_

EchoYJJ, 2020-12-20:
Also, when I run "Win+R", it shows "x64_x84". I don't know whether it's right.

YWangXJ, 2020-12-20:
x64_x86 is 32bit instead of 64bit. Switch to x64 and give it a try. 
[image]

EchoYJJ, 2020-12-21:
@YWangXJ Thanks a lot!

closed 2020-12-31 by EchoYJJ

huatianxue1994, 2021-05-26:
Hello, I meet the same question. And I use the x64 just like follow:
[image]
When I "make launch",there is a error, just like follow:
[image]
What should I do? Thank you advanced!
(PS:,I installed the VS2017)

TedMay, 2021-10-05:
> Hello, I meet the same question. And I use the x64 just like follow: [image] When I "make launch",there is a error, just like follow: [image] What should I do? Thank you advanced! (PS:,I installed the VS2017)

Hello, do you have solution for this problem?

luckysouthchou, 2022-05-01:
same problem here. VS 2019, follwed the newest official install document .

sp2150, 2022-05-23:
Same problem here: 

-- The C compiler identification is MSVC 19.29.30145.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: G:/Visual Studio/2019/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x86/cl.exe
-- Check for working C compiler: G:/Visual Studio/2019/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x86/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.23/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "G:/Visual Studio/2019/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: G:/Carla/Build/zlib-source/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake -f Makefile /nologo cmTC_42db8\fast &&   C:\Program Files (x86)\GnuWin32\bin\make.exe  -f CMakeFiles\cmTC_42db8.dir\build.make /nologo -L                  CMakeFiles\cmTC_42db8.dir\build
    Der Befehl "C:\Program" ist entweder falsch geschrieben oder
    konnte nicht gefunden werden.
    NMAKE : fatal error U1077: "C:\Program": R³ckgabe-Code "0x1"
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

hhf123, 2023-03-01:
I have changed to x64,but there is still this error,do you have another solution for this problem?
请问，问题解决了吗？能告诉我您怎么解决的吗

SandraVict0r, 2023-04-05:
Hi i have the same issue someone has finally resolved this ?

tashby-ARA, 2023-06-22:
I'm also running into this issue.  I have cmake tools installed through visual studio 2019.

cmake version 3.20.21032501-MSVC_2

I am running _x64 Native Tools Command Prompt for VS 2019_ (not the x86_x64 version).

tashby-ARA, 2023-06-22:
To those who still are running into the issue after running x64 Native...

Run "**where mt**" inside the x64 Native Command Prompt for VS.  If it prints "INFO: Could not find files for the given pattern(s)." then there's your problem!  If it prints a valid "mt" file, double check that the path is in the correct Windows SDK version you're targeting.

The fix is to run the batch file "**vcvarsall.bat**" for whatever your VS version is.  You have to pass the parameters in yourself to get the correct version.  If you need help, just run the vcvarsall.bat in your native cmd window without any parameters, it will tell you what you need to do.

After running this with the correct parameters, check "**where mt**" again and verify it is correct now.  Then try compiling Carla again!

chintalapudisarath2508, 2025-04-15:
I have made the following changes and this resolved the zlib issue

C:\Users\sarat\Desktop\Software\CARLA\carla\Util\InstallersWin\install_zlib.bat (Line number: 103)
Update it as shown below
cmake -G "NMake Makefiles"^
 -DCMAKE_INSTALL_PREFIX="%ZLIB_INSTALL_DIR:\=/%"^
 -DCMAKE_BUILD_TYPE=Release^
 "%ZLIB_SRC_DIR:\=/%"
 
Change here is to add this ":\=/" in "%ZLIB_SRC_DIR%" as shown above

C:\carla\Build\zlib-source\CMakeLists.txt (line number: 1)
cmake_minimum_required(VERSION 3.5)

I have made the following changes and rerun make PythonAPI command. The recast issue is resolved

C:\Users\sarat\Desktop\Software\CARLA\carla\Build\recast-src\CMakeLists.txt (Line Number: 1)
cmake_minimum_required(VERSION 3.5)