carla-simulator/carla #3745: Carla 0.9.10 segmentation fault when simulating 1000 cameras and 400 cars.

yangyueren, 2020-12-20:
I am using carla([Ubuntu] CARLA_0.9.10.1.tar.gz) with my own map imported from osm file,  the same osm in #https://github.com/carla-simulator/carla/issues/3686#issue-759431000 

The RGB camera number is 1000, and the vehicle number is 400. I want it to run for 16 minutes, but I got segmentation fault after rendering 2519 frames.

Is there any guide for rendering 1000 cameras?

corkyw10, 2021-03-10:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-15:
I will take a look for issues with many cameras. We have made a number of changes to the OSM to OpenDRIVE converter which may fix the issue. A different matter is the size of the map. Currently Unreal Engine only accepts maps of limited sizes, if the cameras render out of bound it could cause stability issues, do you get segmentation fault on the client alone or is the server crashing as well?

closed 2023-06-02 by stale[bot]