carla-simulator/carla #3746: Pygame window not working properly: "Warning: TF_REPEATED_DATA ignoring data with redundant timestamp"

anavc97, 2020-12-20:
I'm trying to run `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch` but the pygame window does not work properly: sometimes it changes colours to different types of blue, other times it goes all to white or black. 
I've checked that this also happens for the RGB camera sensor and when I run manual_control.py. 
No errors are shown but there are warnings that keep showing. I dropped a screenshot below. From the warnings I assume it has something to do with pygame's buffers so I tried different versions of pygame with no luck. 

Any help would be appreciated. Thanks.

**My setup:**

-  Ubuntu 20.04.1 LTS
- ROS Noetic
- CARLA 0.9.10
- Egg: carla-0.9.10-py3.7-linux-x86_64.egg
- Pygame versions I've tried: 2.0.0, 2.0.0.dev24, 2.0.0.dev22, 2.0.0.dev20, 1.9.6
- carla-ros-bridge installed from source: https://github.com/carla-simulator/ros-bridge.git
- NVIDIA Corporation GM108M [GeForce 940MX]
- Nvidia drivers:  NVIDIA driver metapackage from nvidia-driver-455

[image]

corkyw10, 2021-03-10:
@joel-mb could you have a look at this please?

anavc97, 2021-03-15:
> @joel-mb could you have a look at this please?

Thank you for your time. I've edited my original post with some new information I found out since the post was made.

closed 2023-06-02 by stale[bot]