carla-simulator/carla #3749: writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt zip_safe flag not set; analyzing archive contents... carla.__pycache__.libcarla.cpython-38: module references __file__ creating dist creating 'dist/carla-0.9.10-py3.8-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it removing 'build/bdist.linux-x86_64/egg' (and everything under it) BuildPythonAPI.sh: Success! Traceback (most recent call last):   File "spawn_npc.py", line 26, in <module>     from carla import VehicleLightState as vls ImportError: cannot import name 'VehicleLightState' from 'carla' (../carla/dist/carla-0.9.10-py3.8-linux-x86_64.egg/carla/__init__.py)

KangChou, 2020-12-22:


KangChou, 2020-12-22:
Traceback (most recent call last):
  File "spawn_npc.py", line 26, in <module>
    from carla import VehicleLightState as vls
ImportError: cannot import name 'VehicleLightState' from 'carla' (../carla/dist/carla-0.9.10-py3.8-linux-x86_64.egg/carla/__init__.py)

Shubhamcl, 2020-12-22:
I faced the same error, so I moved to the stable version that is 0.8.2, that seems to work, and has documentation if you select stable version on readthedocs

KangChou, 2020-12-25:
Have you solved this problem now?

corkyw10, 2021-03-10:
Hi @KangChou are you still facing this issue? These problems often arise from using different versions of Python, did you install the API and dependencies with python 3.7?

Datalux, 2021-04-13:
Same issue. Installed carla with Python 3.7. Anyone solved this error?

ajdroid, 2021-04-18:
I'm seeing the same error with py3.8 and Carla 0.9.11 on Windows. My Pythonpath seems to point to the correct egg but it complains 
> ImportError: cannot import name 'VehicleLightState' from 'carla'

closed 2023-06-02 by stale[bot]