carla-simulator/carla #3754: How to Spawn Static Props from BluePrint library in the World

aadilh, 2020-12-23:
I am working on generating street view dataset using CARLA with randomly generated objects like Traffic Signs, Street Signs, etc. but I could not find a way to spawn these in the world using Python API. Is there a way to do this currently or do we plan to include this in future?

XGodina, 2021-01-11:
Hi @aadilh,
As far I know the only way to spawn props is by using the script spawn_npc.py --filterv prop.static.(nameofprop). Did you use it?

aadilh, 2021-03-08:
Hi @XGodina,
I did not use it but from the documentation it seemed like filterv is a filter for vehicles and would spawn objects at vehicle spawn points. Do you know any place where I can read more about it?

closed 2023-06-02 by stale[bot]