carla-simulator/carla #3758: rsync: change_dir "/home/ue4/carla//./Plugins" failed: No such file or directory (2)

will-sloan, 2020-12-28:
I'm trying to build the docker image, and I am at the last step. 
`docker build -t carla -f Carla.Dockerfile .`
But it gives the error below. 
I am on Ubuntu 18.04, and the carla version is whichever is used by the dockerfile. 

I included some lines above for context, not sure if that helps. 
```
Sumo/util/create_sumo_vtypes.py
Sumo/util/netconvert_carla.py
Sumo/util/sequential_types.py
Sumo/util/data/
Sumo/util/data/opendrive_netconvert.typ.xml
rsync: change_dir "/home/ue4/carla//./Plugins" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
make: *** [package] Error 23
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
The command '/bin/sh -c cd /home/ue4 &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-
simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   
cd /home/ue4/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   
make package &&   rm -r /home/ue4/carla/Dist' returned a non-zero code: 2
```

will-sloan, 2020-12-30:
The referenced issue fixed it for me.

closed 2020-12-30 by will-sloan

thillRobot, 2020-12-31:
I am having the same issue. @will-sloan, Can you elaborate on the fix?

will-sloan, 2020-12-31:
@thillRobot My fix was to change line 166 in `Package.sh` in the Docker container to:
`copy_if_changed "./Unreal/CarlaUE4/Plugins/" "${DESTINATION}/Plugins/"`
Here are the steps I followed: 
1 - In `Util/Docker`, create a file called `MyPackage.sh` 
2 - `cd ..`
3 - run `echo BuildTools/Package.sh > Docker/MyPackage.sh`
4 - Change line 166 in `MyPackage.sh` to what I put above.
5 - `cd Docker`
6 - change `Carla.Dockerfile` to: 
```
FROM carla-prerequisites:latest

ARG GIT_BRANCH

USER ue4

RUN cd /home/ue4 && \
  if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git; \
  else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi && \
  cd /home/ue4/carla && \
  ./Update.sh && \
  make CarlaUE4Editor && \
  make PythonAPI && \
  make build.utils
WORKDIR /home/ue4/carla
COPY MyPackage.sh .   # Going off of memory so this may not be how COPY works. But you want to put MyPackage.sh in /carla
RUN echo MyPackage.sh > Util/BuildTools/Package.sh # Update Package.sh
RUN make package
RUN rm -r /home/ue4/carla/Dist

WORKDIR /home/ue4/carla
```
7 - Run `docker build -t carla -f Carla.Dockerfile .`

Hopefully that makes sense.

thillRobot, 2020-12-31:
Thanks for the help. I just applied the fix, and I am trying the build again now. Forgive my ignorance, but in your step 3 `echo` did not copy the contents of `BuildTools/Package.sh` into the new file so I used `cp` instead. Am I misunderstanding something simple?

will-sloan, 2020-12-31:
I might've messed up in the instructions, but you just need to create a copy of `Package.sh` and `cp` works just as well.

thillRobot, 2020-12-31:
Cool. I will let you know if it works. Thanks again.

will-sloan, 2020-12-31:
So I just realized I meant to use `cat`! I made the change to the post above, and please note that the there was a mistake in the Dockerfile which has been fixed.

thillRobot, 2020-12-31:
OK. I have replaced `echo` with `cat` in the Dockerfile, and I am trying again. My connection is not the best, so I have to wait for the 14G download. Is there a way to not repeat the download?

will-sloan, 2020-12-31:
I think you could continue from the last [layer id](https://stackoverflow.com/questions/26220957/how-can-i-inspect-the-file-system-of-a-failed-docker-build) but haven't tried myself.
You might be able to change `Carla.Dockerfile` to:
Replace LASTEST_LAYER_ID with a layer id after the download:
```
FROM LASTEST_LAYER_ID
...
# Put the rest of the file that hasn't yet run
```

thillRobot, 2020-12-31:
I believe I have made progress, but the COPY line in the Dockerfile fails with the following error:

```
Step 6/10 : COPY MyPackage.sh .   # Going off of memory so this may not be how COPY works. But you want to put MyPackage.sh in /carla
COPY failed: file not found in build context or excluded by .dockerignore: stat #: file does not exist
```

The file `Util/Docker/MyPackage.sh` exists so I assume it is a permissions error. I have added execution permissions to the file, and I am trying again.

will-sloan, 2021-01-01:
My bad, its not a permission error. I put the comment with a `#` but this passes it as an argument. 
It would probably work without the comment.

thillRobot, 2021-01-01:
Happy New Year! The build finished successfully. Thank you for all of your help. @will-sloan.

will-sloan, 2021-01-01:
Happy New Year and no problem!

thillRobot, 2021-01-04:
@will-sloan, I am now running into a very similar error when building from source running `make package`. Have you seen this also?

will-sloan, 2021-01-04:
Sorry, I haven't ran 'make package' but if you post the error here or in a new issue, I might be able to help you.

thillRobot, 2021-01-04:
Thanks. Of course there is a possibility that this is related to my environment, but the error is nearly identical to the one shown above. Like a smart guy, I did not save the error or the log info. I am running it again, and I will post it later.

thillRobot, 2021-01-04:
I have learned a little more. I believe that most of the process has completed, but has failed near the end on `Package.sh: Adding extra files to CARLA package.`. I just realized that the package for distribution was created in `carla/Dist`, but not as a **.tar.gz** like I was expecting. I have not tested the package, but there is a **.egg** so I assume it will work.

Either way, I have posted the error below. Also, I am running this on 18.04.
```
make clean && make package
```
Here is the bottom of the error I receive.
```
Sumo/util/data/opendrive_netconvert.typ.xml
rsync: change_dir "/home/thill/carla_simulator/carla//./Plugins" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 23
```
I am interested in the solution, but this is not stopping my progress at the moment.

ARSekkat, 2021-01-04:
I think this change will fix your problem: https://github.com/carla-simulator/carla/commit/7374804542ce1796ddfc022ef91603f526d2cf68#r45498679

will-sloan, 2021-01-05:
Ya that is the same error. Change is the same except docker isn't evolved.

thillRobot, 2021-01-05:
Thanks for your help. I will try the change tonight.

Xiang95, 2021-05-19:
> My bad, its not a permission error. I put the comment with a `#` but this passes it as an argument.
> It would probably work without the comment.
Hello, I just followed you tutorial, but i got the following error:
Can you please help me with it,  thank you in advance.
{
Archiving to
********** ARCHIVE COMMAND COMPLETED **********
BUILD SUCCESSFUL
ERROR: Failed to find command copy_if_changed
       (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=1 (Error_Unknown)
RunUAT ERROR: AutomationTool was unable to run successfully.
make: *** [package] Error 1
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
The command '/bin/sh -c make package' returned a non-zero code: 2

}

PR #3784 (2021-01-13): Make /Plugins/ folder optional when packaging
#### Description

Now when we make the default package (_make package_) it tries to copy the folder **Plugins** to the package, without checking first if it exists (this folder is optional).

This fix checks if the folder exist before copying its content.
Some issues where reported by this using Docker.

Fixes #3758

#### Where has this been tested?

  * **Platform(s):** Windows 10, Ubuntu 18
  * **Python version(s):** -
  * **Unreal Engine version(s):** UE 4.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3784)