carla-simulator/carla #3759: Same vehicle accelerates differently for the same control values in autopilot and manual mode

idilsulo, 2020-12-28:
I have collected data regarding to a vehicle from the `autopilot` mode which contains steering, throttle, brake values for each frame. Now, I am trying to apply the same values manually by using `vehicle.apply_control()`. I am using Carla 9.10 on Ubuntu 18.04.

The strange thing is that the vehicle is accelerating differently in manual mode. 

Below, I have provided a snippet from the logs showing the difference:

```
frame 4
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.0017595547251403332
Vehicle location: (x,y,z): (131.8480682373047,109.40018463134766,0.21403320133686066)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.21427814662456512)
Speed    : 0.00030769422334713095
Data Speed: 0.00017221971939579376

frame 5
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.0018035353859886527
Vehicle location: (x,y,z): (133.6665802001953,109.40318298339844,0.21418368816375732)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.2143104076385498)
Speed.    : 2.308052232178059
Data Speed: 0.00014652180834822726

frame 6
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.001846789033152163
Vehicle location: (x,y,z): (136.4123077392578,109.41195678710938,0.21464145183563232)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.21424682438373566)
Speed.    : 4.357276740169651
Data Speed: 0.00033335725345253506

frame 7
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.0018900427967309952
Vehicle location: (x,y,z): (139.58201599121094,109.42862701416016,0.2145788073539734)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.21407859027385712)
Speed.    : 5.70998158621341
Data Speed: 0.0004186156729710587

frame 8
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.001936749555170536
Vehicle location: (x,y,z): (143.047607421875,109.45508575439453,0.21447443962097168)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.2141101360321045)
Speed.    : 6.218999332016073
Data Speed: 0.0005244400235834583

frame 9
Vehicle - Throttle: 0.699999988079071, Brake: 0.0, Steering: 0.00196419027633965
Vehicle location: (x,y,z): (147.15687561035156,109.4975357055664,0.21449138224124908)
Data location: (x,y,z): (131.6900177001953,109.40003967285156,0.21432361006736755)
Speed.    : 7.198237012039484
Data Speed: 0.001106589334085749
```

The data location and data speed refers to the data collected in autopilot mode. Others correspond to the values after the same steering, brake and throttle values are applied manually.

In manual mode, I am using the same starting position, vehicle and town. Still, the vehicle accelerates differently and achieves a higher speed compared to autopilot mode. What might be the reason behind this? Any help would be appreciated.

idilsulo, 2020-12-28:
I also provide the same gear and hand_brake values. Yet, the problem remains.

corkyw10, 2021-03-10:
@jackbart94 are you able to have a look at this please?

closed 2021-11-18 by jackbart94