carla-simulator/carla #3760: [Feature Request] Server-Side Bounding Boxes

jbeck9, 2020-12-29:
The most common issue that makes the client side bounding box script not so useful is the fact that it can't detect whether the object is visible in scene or not #3601 #1743 ...

This issue is much easier to resolve on the server side with a handful of ray traces per object to check for visibility.

In my opinion, the most intuitive solution is a camera sensor that returns bounding boxes. Given a resolution and FOV, the sensor would return:
*Object type for each actor (vehicle, pedestrian, etc) detected.
*location, orientation, and dimension in 3D world space of all visible objects.
*location and dimension in 2D image space of all visible objects.
*Boolean whether the bounding box is truncated.
*Boolean whether the bounding box is occluded.

The above would allow a fully automated pipeline from simulation to object detection with no wonky segmentation steps in the middle.

I may end up doing this myself, but it is a common enough problem I wanted to reach out and see if anyone else had implemented anything along these lines. Thanks.

PTorr, 2021-01-05:
I was just looking for a solution to this.

You can also use intersection with segmentation map.

corkyw10, 2021-03-10:
@doterop could you have a look at this please?

mittkongg, 2022-03-01:
> I was just looking for a solution to this.
> 
> You can also use intersection with segmentation map.

any results?

Jaivra, 2022-06-06:
any solutions?

closed 2023-06-02 by stale[bot]