carla-simulator/carla #3763: get_up_vector() pointing downwards

paulsans, 2020-12-30:
Hi,

when I call `get_up_vector()` on a zero-Transform or zero-Rotation, the resulting vector points downwards (z=-1):

`Transform().get_up_vector()` gives me `Vector3D(x=-0.000000, y=0.000000, z=-1.000000)`.

According to the definition of the coordinate system, z should be +1.
This is also supported by the transformation matrix:

`Transform.get_matrix()` gives `[[1.0, 0.0, -0.0, 0.0], [0.0, 1.0, 0.0, 0.0], [0.0, -0.0, 1.0, 0.0], [0.0, 0.0, 0.0, 1.0]]`.

Tested in 0.9.10.

corkyw10, 2021-03-10:
@DSantosO could you have a look at this please?

DSantosO, 2021-03-12:
Hello @paulsans,
Indeed, there was a bug in the function that computes the up vector. I have opened the PR #3979 for fixing it. Thank you very much!!

closed 2021-03-12 by DSantosO