carla-simulator/carla #3764: waypoints missing? (IndexError: deque index out of range)

kino3, 2020-12-30:
**Describe the bug**
I'm trying to play the following OpenSCENARIO (`sample.xosc`) which describes intersection (`X-Junction-sample.xodr`). The ego vehicle goes straight and the other vehicle turns left according to the waypoints in the OpenScenario.
[scenario.zip](https://github.com/carla-simulator/scenario_runner/files/5740549/scenario.zip)

However, the ego vehicle suddenly vanished since it cannot find next waypoints (I suppose).

```
Traceback (most recent call last):
  File "scenario_runner.py", line 399, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/shuji/Documents/git/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 133, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/shuji/Documents/git/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 174, in _tick_scenario
    self.scenario_tree.tick_once()
  File "/home/shuji/.local/lib/python3.6/site-packages/py_trees/behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "/home/shuji/.local/lib/python3.6/site-packages/py_trees/composites.py", line 578, in tick
    for node in child.tick():
  File "/home/shuji/.local/lib/python3.6/site-packages/py_trees/behaviour.py", line 249, in tick
    new_status = self.update()
  File "/home/shuji/Documents/git/scenario_runner/srunner/scenariomanager/scenarioatomics/atomic_behaviors.py", line 364, in update
    actor_dict[actor_id].run_step()
  File "/home/shuji/Documents/git/scenario_runner/srunner/scenariomanager/actorcontrols/actor_control.py", line 154, in run_step
    self.control_instance.run_step()
  File "/home/shuji/Documents/git/scenario_runner/srunner/scenariomanager/actorcontrols/npc_vehicle_control.py", line 90, in run_step
    control = self._local_planner.run_step(debug=True)
  File "/home/shuji/Documents/git/carla/PythonAPI/carla/agents/navigation/local_planner.py", line 239, in run_step
    self._compute_next_waypoints(k=100)
  File "/home/shuji/Documents/git/carla/PythonAPI/carla/agents/navigation/local_planner.py", line 177, in _compute_next_waypoints
    last_waypoint = self._waypoints_queue[-1][0]
IndexError: deque index out of range
deque index out of range
Resetting ego-vehicle!
Resetting ego-vehicle!
Destroying ego vehicle 87
ERROR: failed to destroy actor 87 : unable to destroy actor: not found 
No more scenarios .... Exiting
```

**Addtional context**

- The ego vehicle can go straight if I explicitly specify the next waypoints in the OpenSCENARIO (`AssignRouteAction` in the `sample.xosc`, which is commented out Line 125-151). I hope the ego vehicle moves without such an additional action.
- I found the ScenarioRunner cannot find the next waypoints from the connecting road (the road which is not an initial road). Does the OpenDRIVE file contain some bugs?

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version 0.9.10.1
 - Python version 3.6.9
 - Version 0.9.10

glopezdiest, 2021-01-07:
This seems to be related to the OpenDRIVE map, not being properly generated.

By using the controller, the local planner calculates the waypoints using the opendrive map, and as it has some unconnected roads, the local planner fails with IndexError.

However, by using *AssignRouteAction*, you are just manually setting those waypoints, so no calculus in being done. 

I'd recommend you make sure that the road you are at is connected to the next

kino3, 2021-01-13:
@glopezdiest Thank you for your comments. I will re-check the road connection.

closed 2021-07-21 by stale[bot]