carla-simulator/carla #3765: Access to wheels rotation angle in PythonAPI

unumkam, 2020-12-30:
This feature might be helpful for precise position estimation. Coupling position sensors with wheel odometry seems to be more intuitive way to build feedback-based control of car. I was wondered Carla not provides this data.
Locally I have done it through ACarlaWheeledVehicle, then filled ActorDynamicState from WorldObserver and get it accessible from Python Vehicle type

corkyw10, 2021-03-10:
@DSantosO could you follow up on this please?

DSantosO, 2021-03-10:
Hello @unumkam,

This is currently being implemented in PR #3868 and it will be soon in dev branch so I close the issue. Feel free to reopen if necessary.

closed 2021-03-10 by DSantosO

NishanthARao, 2021-03-27:
@DSantosO Hi.

From the PRs, I see that two new functions are added: SetWheelSteerAngle and GetWheelSteerAngle.

However, the question asks the rotation angle of the wheel and not the steer angle of the wheel right? Could we use the GetRotationAngle() that UVehcileWheel provides?