carla-simulator/carla #3766: ARGS="--package package_name" Causing errors when importing

will-sloan, 2020-12-30:
I'm on `Ubuntu 18.04`. 
I was following the [source map ingestion guide](https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source), and the name of package caused an error. 

`make import ARGS="--package carlanoloops"`:

```
parse-options: unrecognized option '--package'
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.10.1-30-g31fff28b-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/sam3sim/UnrealEngine_4.24'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/sam3sim/carla/Unreal/CarlaUE4'
bash "/home/sam3sim/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/sam3sim/UnrealEngine_4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/sam3sim/carla/Unreal/CarlaUE4/CarlaUE4.uproject" --package carlanoloops

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.24/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/sam3sim/UnrealEngine_4.24).
Creating makefile for carlanoloops (no existing makefile)
ERROR: Couldn't find target rules file for target 'carlanoloops' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /home/sam3sim/UnrealEngine_4.24/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/sam3sim/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:83: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 2
```

At the top is says that `package` is an unknown option, but that is what the docs says to do. 

Running `make import` without `package` works, but will only work the first time (as mentioned in the docs).

will-sloan, 2020-12-31:
I tried it on windows and it works. So it might be OS related.

thillRobot, 2021-01-02:
Have you tried `--packages` ? That is what the docker tools uses.

will-sloan, 2021-01-04:
Ah thats probably the problem. Thanks!

will-sloan, 2021-01-04:
Guess I got a bit ahead of myself. Running `make import ARGS="--packages jan4th"` returns the same problem.

thillRobot, 2021-01-04:
It seems like we are working on similar things at the same time. I have not tried that yet because I used `ImportAssets.sh` for the map ingestion. This worked, but the folder structure and **.json** had to be made manually following the older docs for creating a stand alone package. If you are interested I can share.

thillRobot, 2021-01-04:
I have not attempted ingestion in a package from source yet. I was able to ingest a map package using the docker tools, but I had to setup the **.json** and directory structure manually. I can share if you are interested.

will-sloan, 2021-01-04:
Please share, that would be very helpful. I ended up switching to ingestion from source since I was not sure what to do after creating the package so that would be great.

thillRobot, 2021-01-04:
OK. Here are the steps I have taken. I am still learning so I am open to suggestions and/or corrections. 

1. Clone CARLA from github.
2. Build CARLA with docker following the docs [here](https://github.com/carla-simulator/carla/tree/master/Util/Docker) and the fix from #3758.
3. *Prepare package manually* following the docs at bottom of page [here](https://carla.readthedocs.io/en/latest/tuto_A_add_map/). and place the package folder in `./carla/Import/`.  One **.xodr** file and one **.fbx** file are required (**.fbx** may be optional) for each map along with one **.json** file for each map asset package. Here is my directory and **.json** file. 
```
.
+-- Import
|    +-- Package03
|    |    +-- Map03
|    |    |    +-- Map03.fbx
|    |    |    +-- Map03.xodr
|    |    +-- Package03.json 
|    +-- README.md
```

**Package03.json**
```
{
  "maps": [
    {
      "name": "Map03",
      "source": "./Map03/Map03.fbx",
      "use_carla_materials": true,
      "xodr": "./Map03/Map03.xodr"
    }
  ],
  "props": [
  ]
}
```
4. Perform *Map ingestion in a CARLA package* with docker following the same docs at top of page. Create a folder `./carla/Ingested` (I chose that name), and run the command shown below. I used absolute paths in the command, but I am not sure if this is necessary.  
```
cd carla/Util/Docker
./docker_tools.py --input <path to root>/carla/Import --output <path to root>/carla/Ingested --packages Package03
```
If successful, you will see the **.tar.gz** package file in the output directory.
5. Move the file the `carla/Import/` of the package you are going to run. At the point `carla/` cloned in step one has not been built. I used a separate `0.9.11` carla package. Import the package following the docs and you should the asset package appear in `./carla/CarlaUE4/Content/`

I am working on my personal docs for the whole map making process. When they are cleaned up and tested, I would be glad to share those as well.

will-sloan, 2021-01-04:
Thanks for the steps. I went through them but I'm getting an error which I think stems from `docker_tools.py`

This is the output from running: `python3 docker_tools.py --input /home/sam3sim/carla/Import --output /home/sam3sim/carla/MyExport --packages Package01`

```
ue4@carla:latest$ make package ARGS="--packages=Package01"
parse-options: unrecognized option '--packages=Package01'
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: CARLA version 0.9.11-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
parse-options: unrecognized option '--packages=Package01'
BuildCarlaUE4.sh: Using Unreal Engine at '/home/ue4/UnrealEngine'
BuildCarlaUE4.sh: Build CarlaUE4 project.
```

I'm going to try changing `parse-options` to pass the package name as `--packages Package01`, which is what I think it is suppose to be.

corkyw10, 2021-03-10:
@bernatx could you have a look at this please?

jingluoquanergy, 2021-04-08:
make import ARGS="--package=my_package_name" works for me.

jingluoquanergy, 2021-04-08:
correction for the above command:
make import ARGS="--packages=my_package_name"

bernatx, 2021-04-09:
Hi,
The command `make package ARGS="--packages=Package01"` is correct. 
We use similar command like `make package ARGS="--packages=AdditionalMaps"` and it works.
You can check the version of **make** you are using.

I test your command and it worked on Windows machine:
```
$ make package ARGS="--packages=Package01"
-[Package]: [Batch params]: --ue-version 4.26 --packages=Package01
Parsing packages...
Preparing environment for cooking 'Package01'.
Cooking package 'Package01'...
  - prepare
...
```

So, should it be a system configuration problem?

bernatx, 2021-04-09:
I have seen the original problem. There are two things to note here:

1. You need to have the source compiled before you try to import anything.  If you don't have CARLA compiled at the time you import, it will try to compile all needed, and it will pass all parameters through all the building scripts, and some of them will conflict with parameters that are not targeted to that script, like this one:
`bash "/home/sam3sim/UnrealEngine_4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/sam3sim/UnrealEngine_4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/sam3sim/carla/Unreal/CarlaUE4/CarlaUE4.uproject" --package carlanoloops
` The last parameter should not be there, because it makes the build system fail. It is there because the build system propagates the arguments to all scripts.
2. The error `parse-options: unrecognized option '--packages=Package01'` only means that current script does not uses that parameter, but it does not make the script to fail. Other script called by this one could use that parameter, so that is fine. We will try to hide that warning.

So, to fix the problem, first compile the project with
`make PythonAPI CarlaUE4Edtiro`
then use the import command
`make import ARGS="--package carlanoloops"`

I hope that helps.

ShoofLLC, 2021-06-15:
> correction for the above command:
> make import ARGS="--packages=my_package_name"

Had the same error and this command works. I have also built both the editor and the PythonAPI, maybe a configuration issue on my system but I kept having the same problem no matter what I did.

closed 2023-06-02 by stale[bot]

Danlaw-Nikhil, 2024-01-19:
@thillRobot , Were u able to export the Map ? , any supporting documents would be helpful. I have the .fbx,the .xodr, and the .som file with the proper .json file too , But i am unable to generate the Map package , i am suspecting that i did something wrong.

thillRobot, 2024-01-19:
Can you show or explain what you mean by _export the map_? This thread started with _map ingestion_ not exporting. 

I do have some notes about working with CARLA maps on github, but they are a couple years old now. I would refer to the official docs first, and if you have specific questions send them my way or start a thread.

Danlaw-Nikhil, 2024-01-22:
hey  @thillRobot ,
I have followed the procedures in the same way as mentioned in various carla docs.
I have downloaded an osm file from https://www.openstreetmap.org/
created a .fbx file for the same using blender
imported both of these files into unreal engine editor (Carla fork) [ But was unable to import the .xodr file as the import option DID NOT contain a .xodr file import option]
[image]

now i am able to launch the CARLA 
[image]

But when i try accessing through the pythonAPI , and plot the waypoints i get is as : 
The green dots u see are the waypoints i get , but the buildings are far away , i suspect that the buildings have to be moved or rotated , I dont know where exactly i should be doing it , either in blender or the UE4 editor.
[image]

any notes that you have will be helpful , Kindly share .
i used this setup :
Ubuntu 20.04
Unreal engine 4.26
CARLA 0.9.15

any suggestions are welcome.
Thank you

MMNavetty, 2024-02-21:
For those struggling with args names, go to Util/BuildTools, you can either look at import.sh or use directly the python file Import.py