carla-simulator/carla #3767: how to reconnect client in carla v0.9.x

nuomizai, 2021-01-01:
I was trying to build an carla environment wrapper for DRL training, but found that the memory will increase every time I load world using ` self.world = self.client.load_world(town)` (even if for the same town) and I couldn't release the memory by `del self.world`. So I tried to disconnect the carla client instead (to release the memory used by the world). I found the `disconnect` method in carla v0.8.x but no related API for carla0.9.x. Does anyone know how to reconnect the carla client or how to release the memory used by carla world?

varunjammula, 2021-01-01:
You can try reload_world()

corkyw10, 2021-03-10:
@nuomizai were you able to resolve your issue?

nuomizai, 2021-03-18:
> @nuomizai were you able to resolve your issue?

No. But I found that it was not necessary for my environment to reload the world every episode, so I just keep the world entity.

closed 2023-02-10 by nuomizai