carla-simulator/carla #3769: Carla 0.9.11 Make PythonAPI/Make Launch issue with Boost (no error code)

yaadram, 2021-01-03:
Windows 10
Visual Studio 2017
Python 3.7 x64
Carla 0.9.11

- **I used the same configuration to successfully build Carla 0.9.9**. 
- Was successful in compiling the UE2.4 patch for this version.
- Cloned Carla repository and added assests.

When running "make PythonAPI" or "make launch"  , after several stages the script freezes in the file copying sequence of **boost 1.72** . I tried various changes i saw in the issue page and the script always freezes on the same location:
[image]
The final printout is :
 common.copy C:\carla\Build\boost-1.72.0-install\boost-1_72\boost\compatibility\cpp_c_headers\cwctype
C:\carla\Build\boost-1.72.0-source\boost\compatibility\cpp_c_headers\cwctype
        1 file(s) copied.

- I tried waiting for more than half an hour and eventually I need to stop the process (ctrl+c). 

Anything I missed ? Any tips? 

Thank you in advance! 
Yaad Grossman

corkyw10, 2021-03-10:
@bernatx could you have a look at this please?

closed 2023-06-02 by stale[bot]

slugdev, 2023-07-19:
Same thing happens on my build

gutianchang, 2023-08-16:
same problem🤦‍♂️