carla-simulator/carla #3771: Randomness in simulation

emailweixu, 2021-01-04:
I found that the carema images are different between different simulation sessions even though the vehicle is placed at exactly the same location. All other sensors have same reading (imu, radar, etc.) except the camera. I am using the synchronous simulation mode.

Can someone explain what might cause the differences?

The reason I am asking this is that my machine learning agent performs worse during testing than training. So far I have confirmed that all model parameters are checkpointed and loaded correctly (i.e., same input leads to same output). So currently I am suspecting that there might be some systematic differences between different simulation sessions.

Thank you

DSantosO, 2021-01-07:
Hello @emailweixu,
Thank you for reaching out, could you provide some more details about the problem? 
What are exactly the differences between images of different simulations? Are you using the last version? Have you checked that you are getting the images of the camera in the client in the correct order?

emailweixu, 2021-01-07:
Thanks for the response.
I am using CARLA 0.9.9. Here are some details about the simulation. I have four cars each with one camera in the simulation. At the beginning of the simulation, I put them to 4 different locations (same across simulation runs).  The simulation runs in synchronous mode. The camera images are obtained using the callback provided by Sensor.listen() function. I save the four images at the beginning of the simulation for the first simulation and compare them again the images I obtained in the another simulation. The images looks visually similar. But there are some strange differences. For the four images I obtained, the number of different values for each camera are something like 1788, 531, 56134, 37424 (note that there are 3*200*100=60000 values per image, and these numbers change slightly for different simulations). So for the first two cameras, most pixels are exactly same. For the third and fourth camera, there are many are different. But even for camera four, the number of same values is still far more than a random change could have.

emailweixu, 2021-01-08:
Thanks for the response.

I am using CARLA 0.9.9. Here are some details about the simulation. I have four cars each with one camera in the simulation. At the beginning of the simulation, I put them to 4 different locations (same across simulation runs).  The simulation runs in synchronous mode. The camera images are obtained using the callback provided by Sensor.listen() function. I save the four images at the beginning of the simulation for the first simulation and compare them again the images I obtained in the another simulation. The images looks visually similar. But there are some strange differences. For the four images I obtained, the number of different values for each camera are something like 5251, 7182, 9184, 2562 (note that there are 3*200*100=60000 values per image, and these numbers change slightly for different simulations). So most pixels are exactly same.  I noticed that most of the different pixels are on things like trees, but there are some on roads, buildings, etc. My worry is that there might be some session specific randomness which causes my ML agent trained from one session does not perform well on a different session.

Another thing bothering me is that after the call to World.tick(), there is no guarantee that the most sensor data is received by the sensor callback function. Is there an easy way to make sure I can get the latest sensor data after World.tick()? Currently, I have to use condition variable for each sensor to do this. But this is not ideal since some sensor (i.e. Collision) may not receive any data in a step so that I can use this technique on those sensors.

DSantosO, 2021-01-08:
Hello @emailweixu,

First, the last thing, the sensor synchronization is something that, right now, has to be done on the client's side. From your words, I think you have managed but if you want an example there is the `sensor_synchronization.py` script that does exactly this.

The first part is more tricky, I guess that the problem is small differences in the animations of trees or other foliage. Notice that, as you are passing from async to sync modes, the exact moment you are initiating the sync mode is not the same in all the simulations. In the last release, 0.9.11, we have added the possibility to reload the world keeping the synchronous mode ([see here](https://carla.readthedocs.io/en/latest/python_api/#carla.Client.reload_world)). With this, you can set the world in synchronous mode and then, reload the world keeping the settings to start your simulation from the exact same 'moment'. But to do this you need to use the last release.

If you change to the new release, be aware that we have also have improved the [client-server synchronization pipeline](http://carla.org/2020/12/22/release-0.9.11/#synchronous-mode-pipeline-update) so maybe you notice some minor differences.

Let me know if you try this.

emailweixu, 2021-01-08:
Hi, @DSantosO, 

For the issue of sensor synchronization, my understanding is that using locks (or similar mechanism) can handle CameraSensor correctly. But for collision sensor, since we don't know whether it is going to receive any new data, so there is no way to  synchronize its data. Am I right? Does the new [client-server synchronization pipeline](http://carla.org/2020/12/22/release-0.9.11/#synchronous-mode-pipeline-update) you mentioned solve this issue?

For the issue of randomness, do you mean that the simulation is deterministic if reloading with synchronous setting?  I will try to see how it works.

Thanks

Wei

emailweixu, 2021-01-09:
Hi, @DSantosO 

Here's what I got using 0.9.11:

Now the difference for the four images are much less. Out of the four cameras, two are exactly same, the other two has 22 and 54 different values respectively. But there seems no clear pattern of the pixels which are different. They just scatter over the whole image.

However, the simulation speed is much slower. 6-7 fps for 0.9.11. Previously I got about 20 fps for 0.9.9. These are result from four cars in simulation.

Another issue with 0.9.11 is that many camera images are total black or miscolored during collision. This happens no matter whether -opengl flag is used or not.

DSantosO, 2021-01-11:
Hello @emailweixu,

Reloading the simulation in sync mode should give a much more deterministic simulation. There are still some issues that we are investigating from Unreal Engine but definitely is the way to go. 
 - Can you post some examples of the different images?
 - The black/miscolored seems a bug, can you post also some of these miscolored images?
 - The performance drop is somehow expected because we are synchronizing more aggressively in the new pipeline but we are also investigating how we can improve it. Sorry about that.

Thank you.

emailweixu, 2021-01-12:
Hi, @DSantosO,

Here are an example pair of images. You won't notice the difference unless you load them and compare them pixel by pixel.
[image]
[image]

Here is a video showing black/miscolored images:
https://user-images.githubusercontent.com/9756952/104384307-1535f100-54e6-11eb-8291-44ca4e3a5f72.mp4

The performance drop of 0.9.11 seems too dramatic to me. Hope you can fix it asap. Before then, I'll have to stick with the current version.

DSantosO, 2021-01-15:
Hello @emailweixu,

The problem with the images is that Unreal's render has some processing options that are not deterministic so there is not much we can do about these small pixel differences.
About the black/decoloration. Is this in sync mode? Are you using pygame for display? Watching the video, I would say that it has to do more with the fps display than with the Carla simulation itself. Do you have the individual images?

closed 2023-06-02 by stale[bot]