carla-simulator/carla #3772: Multiple GPU for Mutiple Camera Sensor

CMing1994, 2021-01-05:
Is this possible that I set more than 3 camera sensor with very high resolution? if I have more than 3 GPUs, just for sure, I am gonna set a server to do this

jimmyw404, 2021-01-08:
I'm also very interested in this!

Axel1092, 2021-03-15:
Hi all, currently Unreal Engine uses a single GPU for rendering. You could create multiple servers using a GPU each and sync them with the PythonAPI to render specific cameras. We haven't testes this yet as we would like to provide native support for this issue, a feature that is in our road map.

deepalcoholic, 2021-05-28:
@Axel1092 
hi Axel, 
Would you like to share some light on the time plan of the multiple GPU rendering road map ?
thanks a lot!

gouchaoer, 2021-10-11:
> Hi all, currently Unreal Engine uses a single GPU for rendering. You could create multiple servers using a GPU each and sync them with the PythonAPI to render specific cameras. We haven't testes this yet as we would like to provide native support for this issue, a feature that is in our road map.

do you know how to use python api to sync multiple carla instances?

TurtleZhong, 2023-11-18:
> > Hi all, currently Unreal Engine uses a single GPU for rendering. You could create multiple servers using a GPU each and sync them with the PythonAPI to render specific cameras. We haven't testes this yet as we would like to provide native support for this issue, a feature that is in our road map.
> 
> @Axel1092
> 
> do you know how to use python api to sync multiple carla instances?

Hi, Do you have any solutions ?