carla-simulator/carla #3774: Walker animation when teleporting actors via Python API

patmalcolm91, 2021-01-05:
I have an external script which models pedestrian movement (position, speed, and orientation), and I want to be able to use this to control CARLA walker actors. I have tested two approaches so far: disabling physics for the actor and teleporting it using ```Walker.set_transform()```, and using a ```WalkerControl``` object with a direction vector and speed corresponding to the difference between the desired location and the current location. The problem with the latter approach is that due to physics, and possibly some other factors I'm not aware of, the resulting trajectory does not precisely match the one calculated by the script. However, in the former approach, the walking animation is not present, with the walkers simply standing straight up and "gliding" across the ground.

I am looking for a way to precisely control the location and orientation of Walkers, while also being able to enable the walking animation using the python API. Is there currently any way that this could be accomplished?

Note: such a feature would also be very useful for Sumo and VISSIM co-simulation with pedestrians.

patmalcolm91, 2021-07-21:
bumping, as this is still relevant.

LetStarFly, 2022-03-23:
I also want to models pedestrian movement, but spawn_npc_sumo.py not support pedestrain any more. What show I do?