carla-simulator/carla #3775: Map Ingestion does not have stop signs

will-sloan, 2021-01-05:
I'm on Windows 10. Pulled from `master` today so Carla should be `0.9.11`.

The problem:

The map I created in RoadRunner does not contain `BP_Stop` objects when imported into carla, following the [source method](https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source).

The cars follow the correct driving paths but do not stop or yield, even though the junctions where set in RoadRunner.

I believe the problem is related to these warnings I get when using the map I created: 

```
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: Traffic sign 213 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 183 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 184 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 214 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 217 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 187 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 185 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 215 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 218 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 188 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 216 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 186 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 219 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: cannot parse georeference: ''. Using default values. 
WARNING: Traffic sign 213 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 183 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 184 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 214 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 217 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 187 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 185 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 215 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 218 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 188 overlaps a driving lane. Moving out of the road...
WARNING: Traffic sign 216 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
WARNING: Traffic sign 186 overlaps a driving lane. Moving out of the road...
WARNING: Failed to find suitable place for signal.
Nav: failed loading binary
```

I don't have stop sign props, so it seems to be just the maneuvers that are getting in the way.

EsquilliDev, 2021-01-18:
Were you able to find a workaround? I'm using the same method to ingest a custom map, but the stop sign are not getting imported.

will-sloan, 2021-01-19:
I found that only some types worked. Like a four way stop with the props chosen automatically by RoadRunner. It seems to work when signs are on the border of the "road" (on the outline that is shown when hovering over the road). Otherwise No.

EsquilliDev, 2021-01-19:
I am getting those warnings now, even after deleting the traffic signs (stop signs). I will update this issue if I found something useful.

Setup:
Ubuntu 20.04
CARLA 0.9.10

EsquilliDev, 2021-01-19:
Update: Found the issue. In my case, the signals are of type "StopLine". If you look inside the .xodr file for those traffic sign ID you will find them there. My stop lines are placed outside the intersections, but in the road. I don't know why would that be a conflict, but that's the culprit.

will-sloan, 2021-01-30:
@EsquilliDev Did you find a solution to the issue you found?

EsquilliDev, 2021-01-30:
So, the actual issue were the stop lines that I placed on RoadRunner. I tried placing them inside/outside the junctions, but it never worked. My fix was to remove the stop lines and just put stop line markings. Then, in CARLA, I used the BP_Stop on each stop line around the map. It looks like this:

[image]

In that image above I'm using two blueprints: BP_Stop and BP_TurnBasedStop.

Note: The stop lines will no longer be in the xodr file.

Axel1092, 2021-03-16:
Hi all, stop and yield signs are generated from the OpenDRIVE file or use existing sings if they are placed directly in the level in the editor. RoadRunner will only export them into the OpenDRIVE file if they are generated using their tools to create traffic sings and link them with the corresponding junction. Notice that in RoadRunner you can place a stop sign anywhere but only define its effect within a junction.

will-sloan, 2021-03-19:
Since I initially opened the issue, I have been able to generating my own signs by selecting `Automatically Place Selected Prop`, so RoadRunner is placing the stop signs. The cars are also following the signals I am setting in road runner. 
The issue is that assemblies (like stop signs, buildings, traffic lights) either get placed in the incorrect spot, duplicated, or not fully rendered. 
Here are some photos of what I mean:

[image]

[image]
[image]

I'm not too sure how much of this is connected to Carla (if at all), but any advice for what to do would be appreciated.

will-sloan, 2021-03-20:
So it seems that the duplicates are due to Unreal Engine adding props then adding props from the xodr file. I noticed this when stop signs and such showed up when I didn't add them in RoadRunner.

Axel1092, 2021-04-13:
@will-sloan, Unreal will add the traffic signals and lights that are present in the xodr file exported by roadrunner. Road runner allows you to customize junctions to place signals as you need but as you see sometimes this information is exported to the xodr in a weird way specially when using their automated tools. My advice is to customize the signals in RoadRunner, make sure that the signal is affecting all lanes that you need (you can see this by selecting the signal). The signals in the middle of the road should disappear if you have a specific signals affecting all corresponding lanes.

axdvgnj153, 2022-02-09:
Hi @ucbmpc2021 
I have the same problem as you. Have any idea to solve it ? thank you

my env :
Carla 0.9.13 with UE-4.26 
RoadRunner  R2021b

ucbmpc2021, 2022-02-21:
Hi @axdvgnj153 

No I haven't solved it yet. I used exactly the same version as you: Carla 0.9.13 with UE-4.26 and Roadrunner R2021b with Ubuntu 18.04.6. Here are some pictures of the duplicated traffic signs&lights in custom map:
[image]

[image]

[image]

closed 2023-06-02 by stale[bot]

drsftx73, 2023-10-31:
This issue has been happening since Carla version 0.9.9. It is related to the fact that Carla does not parse the xodr file properly and purely instantiates a Carla signal wherever there is mention of a signal in xodr. 
In order to fix this, the Carla team would need to support more complex traffic signal configurations. Currently there is no support for protected turns, nor is there support for split phase signalization phases. 
the same thing happens for stop and yield signs

Zhouqianqian, 2024-01-30:
> Hi @axdvgnj153
> 
> No I haven't solved it yet. I used exactly the same version as you: Carla 0.9.13 with UE-4.26 and Roadrunner R2021b with Ubuntu 18.04.6. Here are some pictures of the duplicated traffic signs&lights in my custom map: [image]
> 
> [image]
> 
> [image]
> 
> I would be happy to discuss our problems!

Hi, would you please share how did you import custom map into packaged version
of carla? I have stuck in this problem for over a week, please~