carla-simulator/carla #3777: CarSim problem：no collision

JarvanI, 2021-01-06:
I built 0.9.11 on win10 and generated the .sim file according to the requirements of the official documentation. Then I run manual_control_carsim.py and pressed ctrl+k to enable carsim. But I found that the collision disappeared after turning on carsim, the vehicle could drive into the walls and the house, and even opened the map. The accelerator and brake of the vehicle can be controlled normally, but the steering angle is very small compared to the original, and there is basically no way to steer normally. 
[image]
[image]
I want to know where this problem is my setting wrong? I generated the sim file according to the picture below
[image]

corkyw10, 2021-03-10:
@bernatx could you have a look at this please?

wittenator, 2021-10-20:
Our team is thinking about investing in CarSim to use it in Carla. Is this still a problem in the newest release, @JarvanI ?

closed 2023-06-02 by stale[bot]